microsoft/Azure-Kinect-Sensor-SDK #499: k4a_playback_open issue

darahan, 2019-07-15:
Hi I get the following issue when running my code over k4a_playback_open(filePath, &playback_handle) :

[2019-07-15 13:26:33.310] [error] [t=144] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\record\sdk\playback.cpp (49): k4a_playback_open(). Unable to open file 'recording1.mkv': ios_base::failbit set: iostream stream error
Opened device: 000050292412
Failed to open recording

I checked the file write permssions. I am using Windows 10 and Qt 5.12.0 with MiniGW 64-bit. Here is my code:  

```c++
#include <iostream>
#include <k4a/k4a.h>
#include <k4arecord/k4arecord_export.h>
#include <k4arecord/playback.h>
#include <k4arecord/record.h>

int main()
{
    uint32_t deviceCount = k4a_device_get_installed_count();
    std::cout << "Number of Azure Kinect(s) connected: " <<  deviceCount << std::endl;
    k4a_device_t devices[deviceCount];
    k4a_device_configuration_t deviceConfigs[deviceCount];

    //loop for opening devices:
    for (uint32_t i = 0; i < deviceCount; i++){
        k4a_device_open(i, &(devices[i]));
        // Get the size of the serial number
        size_t serial_size = 0;
        k4a_device_get_serialnum(devices[i], nullptr, &serial_size);

        // Allocate memory for the serial, then acquire it
        char *serial = (char*)(malloc(serial_size));
        k4a_device_get_serialnum(devices[i], serial, &serial_size);
        printf("Opened device: %s\n", serial);

//        if (i == deviceCount - 1){
//             deviceConfigs[i].wired_sync_mode = K4A_WIRED_SYNC_MODE_MASTER;
//             printf("Device: %s\n", serial, " is assigned as a master device");

//        } else {
//            deviceConfigs[i].wired_sync_mode = K4A_WIRED_SYNC_MODE_SUBORDINATE;
//            printf("Device: %s\n", serial, " is assigned as a slave device");

//        }

        deviceConfigs[i].wired_sync_mode = K4A_WIRED_SYNC_MODE_STANDALONE; //To be commented out for the if else code above

        deviceConfigs[i].camera_fps       = K4A_FRAMES_PER_SECOND_5;
        deviceConfigs[i].color_format     = K4A_IMAGE_FORMAT_COLOR_BGRA32;
        deviceConfigs[i].color_resolution = K4A_COLOR_RESOLUTION_3072P;
        deviceConfigs[i].depth_mode = K4A_DEPTH_MODE_WFOV_UNBINNED;
        free(serial);
        // Start the camera with the given configuration
        if (K4A_FAILED(k4a_device_start_cameras(devices[i], &(deviceConfigs[i]))))
           {
               printf("Failed to start cameras!\n");
               k4a_device_close(devices[i]);
               return 1;
           }

    }
    k4a_playback_t playback_handle = nullptr;
    const char* filePath = "C:\\Users\\mbroekens\\Documents\\build-AzureKinect-Desktop_Qt_5_12_0_MinGW_64_bit-Debug\\recording1.mkv";
    if (k4a_playback_open(filePath, &playback_handle) != K4A_RESULT_SUCCEEDED)
    {
        printf("Failed to open recording\n");
        return 1;
    }

    uint64_t recording_length = k4a_playback_get_last_timestamp_usec(playback_handle);
    printf("Recording is %lld seconds long\n", recording_length / 1000000);
    k4a_capture_t capture = NULL;
    k4a_stream_result_t result = K4A_STREAM_RESULT_SUCCEEDED;
    while (result == K4A_STREAM_RESULT_SUCCEEDED)
    {
        result = k4a_playback_get_next_capture(playback_handle, &capture);
        if (result == K4A_STREAM_RESULT_SUCCEEDED)
        {
            // Process capture here
            k4a_capture_release(capture);
        }
        else if (result == K4A_STREAM_RESULT_EOF)
        {
            // End of file reached
            break;
        }
    }
    if (result == K4A_STREAM_RESULT_FAILED)
    {
        printf("Failed to read entire recording\n");
        return 1;
    }

    do
     {

      std::cout << '\n' << "Press a key to terminate...";

     } while (std::cin.get() != '\n');

    //loop for closing devices:
    for (int i = 0; i < deviceCount; i++){
      k4a_playback_close(playback_handle);
      k4a_device_stop_cameras(devices[i]);
      k4a_device_close(devices[i]);
    }
    return 0;
}
```

billpottle, 2019-07-15:
I had this same error when running a similar program which also uses k4a_playback_open(filePath, &playback_handle) from within visual studio on windows 10. However, if navigate through the filesystem and run the program directly from the folder, it loads and plays the file correctly.

xthexder, 2019-07-15:
Under the hood, `k4a_playback_open` just uses `std::fstream.open()`, so any path that works with fstream should work fine with k4a.

Unless you edited the error message to shorten it, your program was trying to open `recording1.mkv` without a path, which will look in the current working directory.
The code you posted should use the absolute path and will work when run from any directory.

Let me know if you're still seeing this issue after checking the path / working directory.
If you want to check file permissions, you should be able to open the same recording with Azure Kinect Viewer to test.

billpottle, 2019-07-15:
> Under the hood, k4a_playback_open just uses std::fstream.open(), so any path that works with fstream should work fine with k4a.  @xthexder - The weird thing with my error is that the file was in the local directory both times. So when I run from visual studio, I get this error. When I click on the file from explorer, no error. This persists over many different builds. I know I'm not using two different files, because the timestamp keeps updating on each build.

xthexder, 2019-07-15:
@billpottle  In your case, it's likely that Visual Studio is running the program from a different working directory. The working directory and the directory the executable are in are not guaranteed to be the same.
By default Visual Studio will run the executable from the project directory, not the build directory.

darahan, 2019-07-16:
I made a mistake in my working directory. I thought the "playback_open " was there to open a file for a new recording instead of opening an existing file. My bad.

closed 2019-07-16 by darahan

billpottle, 2019-07-17:
@xthexder  - Thank you, that is helpful to know. I tried putting the mkv file in the project directory and it opened without issue.