microsoft/Azure-Kinect-Sensor-SDK #500: Python Reference for Microsoft Azure Kinect Sensor SDK

MrudulaSatya, 2019-07-15:
**Is your feature request related to a problem? Please describe.**

Most Machine learning and AI solutions are coded in Python as it has a collection of numerous libraries to do data pre-processing and apply machine learning algorithms.  If the Microsoft Azure Kinect Sensor SDK provides a reference in Python, it will be very handy to use Azure Kinect DK in our AI  applications. 

**Describe the solution you'd like**

A Python Reference that provides functions to access the sensors and cameras on Kinect DK.

**Describe alternatives you've considered**

Right now we have to use C/C++ to obtain the input from Kinect DK and send that obtained input to the python code via sockets/pipelines for use. This creates a latency issue unacceptable for realtime applications.

wes-b, 2019-07-15:
We have discussed Python in the past as another language extension, but so far we have not had time. Check out https://feedback.azure.com/forums/920053 for how we are tracking larger feature requests, we are planning to use that info to help steer our team toward the next features to invest in.

closed 2019-07-15 by wes-b

brendandburns, 2019-09-16:
If anyone comes across this issue, I've started work on a Python binding for the SDK here:

https://github.com/brendandburns/py-k4a

It can currently capture images successfully.

ibaiGorordo, 2020-07-06:
Sorry for commenting in a closed issue.

I have also develop a repository with **Python** programs for the **Azure Kinect DK SDK**. I have examples for _color image_, _depth_ , _IMU_ and more:
https://github.com/ibaiGorordo/pyKinectAzure

Example of smoothed depth image using **OpenCV** in **Python**:

[image]

jrterven, 2020-10-26:
Here's another Python wrapper that I have been working on for the past months. It can get all the image sources, point cloud, and body tracking.
https://github.com/jrterven/KinZ-Python

danfairs, 2021-04-08:
Amazing! Thanks for your work on this. Do you know what release this might find its way into?

MehranTaghipour, 2021-11-25:
hello everyone, I am working on 3d photography by using Azure kinect camera. I have a problem with Intrinsic and Extrinsic parameters estimation in python. I suppose that I need python calibration code, Is there anybody to help me?

PR #1483 (2021-03-24): Python api development

## Fixes #500 
See also [Python support](https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/37675798-python-support).

### Description of the changes:
- Adding Python k4a package with similar API as C++ and C# wrappers using ctypes as bindings to the C API. The bindings are for the API in the SDK v1.4.1.
- Adding functional tests to test Python k4a package.
- Example code is provided as a guide for using the Python k4a package.
- Build scripts is provided to package into wheel file which can be released as part of the SDK.
- Doxygen creates html docume [... cut]