microsoft/Azure-Kinect-Sensor-SDK #501: Runtime Error - multi-camera recording - Ubuntu 18.04 - SDK 1.1

somacoder, 2019-07-15:
The following is after applying the libusb memory fix at: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/485

Both cameras work with individual recordings.

subordinate:
```
$ k4arecorder --device 1 --external-sync subordinate output-2.mkv
Device serial number: 000015192412
Device version: Rel; C: 1.6.98; D: 1.6.70[6109.7]; A: 1.6.14
Device started
[subordinate mode] Waiting for signal from master
```
master:
```
$ k4arecorder --device 0 --external-sync master output-1.mkv
Device serial number: 000019492412
Device version: Rel; C: 1.6.98; D: 1.6.70[6109.7]; A: 1.6.14
[2019-07-15 11:28:31.205] [error] [t=4097] /home/vsts/work/1/s/extern/Azure-Kinect-Sensor-SDK/src/color/uvc_camerareader.cpp (153): Start(). Failed to start streaming: Unknown error
[2019-07-15 11:28:31.205] [error] [t=4097] /home/vsts/work/1/s/extern/Azure-Kinect-Sensor-SDK/src/color/color.cpp (194): color->m_spCameraReader->Start(width, height, fps, config->color_format, &color_capture_available, color) returned failure in color_start()
[2019-07-15 11:28:31.205] [error] [t=4097] /home/vsts/work/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (725): color_start(device->color, config) returned failure in k4a_device_start_cameras()
Runtime error: k4a_device_start_cameras(device, device_config) returned 1
```

System: NUC8i7HVK w/ 32 GB RAM
OS: Ubuntu 18.04, Fully Updated
SDK: 1.1.0

wes-b, 2019-07-15:
Unfortunately that log is not too helpful. Can you rebuild the SDK in debug mode and retry? Use the CMake option -DCMAKE_BUILD_TYPE=Debug.

somacoder, 2019-07-15:
Sure! Not much in those original logs you're right. See attached for equivalent debug logs.

[master.txt](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/3393460/master.txt)
[subordinate.txt](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/3393461/subordinate.txt)

somacoder, 2019-07-15:
Possibly a library is missing for handling the image type?

wes-b, 2019-07-15:
I don't think so and we have not seen this error before. Would you mind updating ```uvc_strerror``` function in your extern libuvc project to convert all the known error codes to strings so we can figure out what error code is comming back?

wes-b, 2019-07-22:
@somacoder any update on this? Any chance you got it working?

closed 2019-07-22 by wes-b

somacoder, 2019-08-12:
Hey @wes-b ! I apologize for the delayed followup here. I was deep into Kinect programming and configuration in Windows.

After reinstalling a clean 18.04, this problem appears to have went away completely! Excelsior!