microsoft/Azure-Kinect-Sensor-SDK #502: Coordinate Frame Transformations

rfilkov, 2019-07-15:
**Is your feature request related to a problem? Please describe.**

Currently k4a_transformation_depth_image_to_color_camera()-function transforms the depth image to the color camera space, and k4a_transformation_color_image_to_depth_camera()-function transforms the color camera image to the depth camera space. There is currently no way though to transform the body index image (returned by the body tracking SDK) to the color camera space and vice versa - the color camera image to the body index space.

**Describe the solution you'd like**

Previously, in Kinect SDK 2.0 there were methods like MapDepthFrameToColorSpace() and MapColorFrameToDepthSpace() that converted the depth frame coordinates to color frame coordinates, and color frame coordinates to depth frame coordinates respectively. It would be good to have similar functions again. This way we could use these coordinate transformation frames for custom frame (e.g. body index frame) transformations, as well.

**Describe alternatives you've considered**

An alternative would be to add the body-index transformation functions to the K4A SDK or to the body tracking SDK, but the coordinate conversion functions suggested above would be more general, I think.

**Additional context**

rabbitdaxi, 2019-07-15:
@rfilkov thank you for opening this issue, there is actually work in progress regarding a related issue #260 that can address similar problem.

rabbitdaxi, 2019-07-15:
Duplicate this issue to #260

closed 2019-07-15 by rabbitdaxi