microsoft/Azure-Kinect-Sensor-SDK #503: k4aviewer.exe doesn't launch in a remote desktop session and doesn't give any error message

this-username-is-taken, 2019-07-15:
**Describe the bug**
k4aviewer.exe doesn't launch in a remote desktop session and doesn't give any error message

**Desktop (please complete the following information):**
 - OS with Version: Windows 1809 build 17763
 - SDK Version:  tried both 1.0.5 and a build from source on branch origin/release/1.1.x

**Additional context**

wes-b, 2019-07-18:
This seems to be failing before k4a_device_open is called as there are no logs

williampMSFT, 2019-07-18:
> This seems to be failing before k4a_device_open is called as there are no logs

Yeah, this happens because you can't get an OpenGL context from an RDP session, so we fail to init the viewer and exit.

I think the fact that it doesn't work is by design (or at least, is a consequence of limitations in OpenGL+RDP), but the fact that the messaging is bad is something we might be able to improve on.

Chris45215, 2019-09-19:
I have encountered this problem as well; it is very bad when the Kinect was advertised for multi-camera use, but that use requires a full desktop PC setup (or high-end gaming laptop) for each and every camera. I would suggest that the developers make a bespoke and clear error message for this problem if possible, because otherwise the troubleshooting process can be very misleading.

The problem gets worse: You may want to try a different type of remote desktop program, especially one that interfaces at a later stage in the pipeline - VNC works that way. But some, and perhaps all, VNC-based remote programs won't work either, unless the remote computer has a monitor attached. This is not an Azure Kinect issue; instead, Windows no longer renders any output if it does not detect an attached screen, and VNC is not detected as a screen. 

The workaround is to buy an inexpensive "display emulator" which you can plug into one of the monitor ports. Then you can use a program like FreeVNC to control the system remotely.

artus-LYTiQ, 2020-11-12:
Might help, might not, but you can start the k4aviewer when logged in with a real console. And then you can transfer your session to RDP by logging in as the same user. It does not *launch* in RDP, but it can be used in RDP if launched before. Thus, auto login at restart would enable your use case. Not a nice work around, though.