microsoft/Azure-Kinect-Sensor-SDK #507: tewrapper_create fails to initialize on linux due to GLXBadFBConfig

xthexder, 2019-07-16:
**Describe the bug**
During initialization of the transformation engine, an OpenGL error occurs when using Nouveau video drivers on Ubuntu.

**To Reproduce**
1. Open k4aviewer (must include #505 bug fix) on Ubuntu
2. Open a device or recording, and switch to 3D point cloud view
3. k4aviewer will hang / crash when initialize the transformation engine dll

**Expected behavior**
The point cloud view should display the same as on windows.

**Logs**
```
X Error of failed request: GLXBadFBConfig
  Major opcode of failed request: 155 (GLX)
  Minor opcode of failed request: 34 ()
  Serial number of failed request: 29
  Current serial number of output stream: 28
```

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04 with latest Nouveau driver
 - SDK Version:  develop branch with #505 fix
 - Firmware version:  N/A, recording file was used.

**Additional context**
I'm unable to debug this any further without the depth engine source.
The GL error occurs in `transform_engine_create_and_initialize`, and the function never returns.

rabbitdaxi, 2019-07-16:
This error should be related to the minimum required OpenGL version. #495 is also tracking it.

Meanwhile, can you use glxinfo | grep “OpenGL” to check whether the OpenGL core profile version >= 4.4? If not, update your GPU latest driver can normally solve this issue

cdedmonds, 2019-07-16:
We are looking at improving the error message and documentation as mentioned in #495 .

closed 2019-07-16 by cdedmonds