microsoft/Azure-Kinect-Sensor-SDK #509: Document API for depth, transform engines and raw modulated IR image format

staniliev, 2019-07-16:
Provide API documentation for depth and transform engines. Document format of raw modulated IR image captured by "capture_get_ir_image" function (which is invoked by depth_engine_thread).
New documentation can be used to implement an alternative CPU depth engine made of portable open source C++ code. This will drop dependency on GPU/OpenGL and will allow to run SDK on a wide variety of ARM SBC with any Linux distribution.

wes-b, 2019-07-16:
Sorry, this is an area of the technology that we as a company are not yet ready to share.

closed 2019-07-16 by wes-b

rabbitdaxi, 2019-07-16:
@staniliev add some more comments here. First, let us separate the transform engine and depth engine. 

- For transform engine:
Transform engine CPU reference code is fully open sourced in the current GitHub. You can see TRANSFORM_ENABLE_GPU_OPTIMIZATION in (https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/src/sdk/k4a.c), currently default is set to true, if you set to false, then the SDK will use the reference C++ code to do transformation (which will be much slower). The reason we have depth engine and transform engine to use GPU is for real time performance consideration. We also have the algorithm explain the functions https://docs.microsoft.com/en-us/azure/Kinect-dk/use-image-transformation

- For depth engine
As Wes mentioned, the depth engine (the time of flight raw to depth algorithm code) is not open sourced at this moment, you can see the license. We have documentation https://docs.microsoft.com/en-us/azure/Kinect-dk/depth-camera explain the concepts. For the dev kit, we leverage GPU for real time performance. There is a already one user voice about ARM support request https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/37615903-arm-support-for-linux

fishcu, 2019-08-23:
@rabbitdaxi What's the license for the closed source depth engine? I mean the `libdepthengine.so.*` or `*.dll` files.

yan796, 2019-10-18:
Hi, I understand that  the depth engine is not open source. However, can we get the 4 correlation measurements per frequency (the time of flight raw) to generate the depth ourself from the dev kit ? This will be super useful...