microsoft/Azure-Kinect-Sensor-SDK #510: No IR data when in passive IR mode

darahan, 2019-07-16:
I was making some images in the WFOV_UNBINNED mode, but found the IR pictures quite overexposed. then I tried to use the PASSIVE_IR mode while using the same settings. The command line argument is below: 

PS C:\Program Files\Azure Kinect SDK v1.1.0\tools> .\k4arecorder.exe -d PASSIVE_IR -c 1080p -r 5 -l 900 --imu OFF test5.mkv

However, if I open the above file in k4aviewer, it says "Depth mode: (None)". Also, it does not display the IR recordings. I was able to identify and extract the b16g file using  MKVToolNix, but I am currently not able to open it. Unfortunately there was no depth track present in the recording. Now my questions are:
-Can files recorded in the PASSIVE_IR mode be fully recongnized by k4aviewer?
-Is it possible to generate a depth track from the above data?
- How can I open and playback the depth recordings?
Cheers

wes-b, 2019-07-16:
Thanks for the issue, this is bug in K4aViewer.

darahan, 2019-07-16:
Yes partially. I did expect a Depth track in passive IR mode however.

wes-b, 2019-07-16:
When using Passive IR mode, then depth is disabled and there is no track to display. In this case there should only be a passive IR track

wes-b, 2019-07-16:
Sorry, I just noticed the last part of your question. We expect the other 4 depth modes to record and playback 2 tracks; Depth and IR, and 1 color track if enabled. They should work with the recorder.exe and k4aViewer.exe

wes-b, 2019-07-16:
Duplicate of #424

closed 2019-07-16 by wes-b

darahan, 2019-07-16:
Ok this still leaves me with the question, why is there no depth  view in --depth-mode in PASSIVE_IR? I find this very counterintuitive naming. I selected this mode due to very overexposed subjects in WFOV and extermely overexposed subjects in NFOV, in both cases plants in a field. So I would love to have a depth view in Passive IR mode to get a better image of these subjects.

wes-b, 2019-07-17:
Ahh, I see what you are saying. How can the ```--depth-mode``` setting, what ever it is, not give me depth data. We have a couple articles that talk about this, linked below. In summary there are 4 modes that turn on the lasers and actively measure time-of-flight, these modes result in a depth image. All 5 modes return in IR only image, and in the case of PASSIVE_IR the image is the ambient IR of the scene with the IR emiters turned off.

https://review.docs.microsoft.com/en-us/azurekinect/azure-kinect-devkit?branch=master#depth-camera-supported-operating-modes
https://docs.microsoft.com/en-us/azure/kinect-dk/depth-camera