microsoft/Azure-Kinect-Sensor-SDK #511: C# | transformation.DepthImageToPointCloud

djambo, 2019-07-16:
**Describe the bug**

I am currently trying to use: **transformation.DepthImageToPointClou**d in a Unity C# script, but the function returns an error: **Unable to allocate Short3 array for format Custom**

**Expected behavior**

In case this is already been fully implemented, I would expect it to return a ArrayImage \<Short3\> containing the 3d coordinates of the cloud vertices.

**Logs**

AzureKinectException: Unable to allocate Short3 array for format Custom

Microsoft.Azure.Kinect.Sensor.ArrayImage`1[T].CreateHandle (Microsoft.Azure.Kinect.Sensor.ImageFormat format, System.Int32 width_pixels, System.Int32 height_pixels, T[]& data) (at /Image.cs:31)

Microsoft.Azure.Kinect.Sensor.ArrayImage`1[T]..ctor (Microsoft.Azure.Kinect.Sensor.ImageFormat format, System.Int32 width_pixels, System.Int32 height_pixels) (at /Image.cs:78)

Brent-A, 2019-07-16:
Thanks @djambo! This functionality is not yet fully implemented. We're working on changing the way the native memory is exposed which will likely mean removing the ArrayImage class in favor of a more generic Image class that can handle both managed and native buffers.

To work around this for now you can try adding this to the switch statement in ArrayImage<T>.CreateHandle

```C#
case ImageFormat.Custom:
    pixelSize = Marshal.Sizeof(typeof(T));
    break;
```

djambo, 2019-07-16:
Thank you so much @Brent-A! This made it work. It's enough to get the list of points into my unity scene.

closed 2019-07-16 by djambo