microsoft/Azure-Kinect-Sensor-SDK #514: Body tracking is too slow and inaccurate

rfilkov, 2019-07-17:
**Describe the bug**

By all means Azure Kinect is the best Kinect so far, and will be probably the best depth sensor on the market. The sensor SDK is pretty stable and good, providing almost everything an average user would want. But the body tracking subsystem is ruining this positive user experience. In means of API this SDK is great too, but the DNN model performance is much worse than the body tracking of Kinect-v2. The joint positions are inaccurate by fast movements. The body index map is not very accurate, as well. It does not fully match the user's silhouette on the depth frame. On my GTX 1060 it takes 2-3 depth frame cycles to process a body frame. Hence, it works at about 10 fps.

**To Reproduce**

1. Run Azure Kinect Body Tracking Viewer.
2. Stand in front of the sensor.
3. Make fast arm movements.
4. Look at the arm joint positions with regard to the real arms.
5. Look at the colorized body-index map with regard to the real body.

**Expected behavior**

1. I expect the body tracking to work at least at 30 fps or more, i.e. faster than the depth frames arrive.
2. I expect the body joint positions to match as precisely as possible the user's joints on the depth frame.
3. I expect the body index map to match as precisely as possible the user's silhouette on the depth frame.
4. I expect the body tracking to be less demanding, in means of hardware and 3rd party software requirements. GTX 1070 + CUDA + cuDNN + manually setting paths would be too much for the average user.

Please consider at least providing some option to the users, who don't have high end graphics cards and would like to get Kinect body tracking out of the box, without (or with minimum) extra installations. As far as I remember, Kinect-v2 used random forest model(s) for its body tracking. The performance was great and no extra installations were needed, back then in 2013/14. 

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  

 - OS with Version: Windows 1809
 - Kinect SDK Version:  1.1.0
 - Body tracking SDK Version:  0.9.0

**Additional context**

I believe most Kinect users would expect better, more accurate and more performant body tracking experience, not worse. And now, with Apple adding people segmentation and body tracking to their AR-Kit 3.0 I would expect Kinect (with all these years of experience) to provide a better user experience in all aspects than anybody else.

billpottle, 2019-07-17:
I understand that the body tracking is only a preview release, and will likely be improved over time. As a stopgap, if it's possible, perhaps the body tracking can expose a parameter that can be tuned to achieve higher accuracy for offline or remote processing on powerful systems for non-real time applications. 

There are many use cases that might only need a 3-4 second clip and could afford to spend minutes/hours processing, but can't afford meter+ gaps between actual and reported joint positions.

Brekel, 2019-07-17:
Fully agree!

I have a feeling like the output data is also overly filtered/smoothed somewhere in the pipeline, possibly at the model fitting stage after the DNN?
Maybe for certain use cases it could be useful to favor plausible poses over per-frame accuracy, but a user setting to tune this behavior could definitely improve usability in many more scenarios.

Another addition could be to output the 2D joint positions straight from the DNN when we need accurate points on the IR frame prior to any model fitting and post filtering stages.

I've had these 2 suggestions on the feedback page already in case you want to upvote:
https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/38027029-user-adjustable-skeleton-smoothing
https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/38031535-2d-joint-positions

cdedmonds, 2019-07-17:
Thank you for the feedback. Please note that the Body Tracking SDK is currently a preview release and we are actively working on many aspects of quality. The installation process and dependencies are a pain point that we are aware of and are working to fix.

As of the preview release the minimum recommended card is a GTX107:
https://docs.microsoft.com/en-us/azure/kinect-dk/system-requirements

Please keep sending us your feedback as new releases come out.

rfilkov, 2019-07-17:
@Brekel I upvoted your first suggestion some days ago and now - the 2nd one as well. Anyway, I don't think a single hyperparameter tuning would solve the lag-issue. DNNs are notoriously slow, even on high end graphics processors.

Brekel, 2019-07-17:
Thanks for the upvote.
Maybe you're right but I think this is not lag/slowness but filtering I'm seeing on my GTX1080ti, could be wrong.

In any case the DNN is still amazingly robust so not a critique just trying to help the team in making it more awesome and useful for more scenarios. The more controls we have the easier that will be.

rfilkov, 2019-07-17:
@cdedmonds Thank you for the swift response! I know the system requirements and that BT SDK is still in preview. But I think it would be better (for both - you and us) to give such a feedback now, while in preview than when the official release is out.

I suppose the Azure Kinect-team is much different than the team developed the Kinect-v2 sensor and SDK, but anyway, you should have access to all the documentation and internal info from back then. It may be worth comparing the K4A body tracking to the body tracking of Kinect-v2, in means of performance, accuracy, features, extras, etc. It would be good to get a well performing, more accurate and more versatile body tracking in the end than the previous one.

Please keep this issue open, so we could comment when the new BT SDK releases come out. When do you plan to release the next preview version, by the way?

cdedmonds, 2019-07-17:
We appreciate the feedback. Keep it coming.

Brekel, 2019-07-17:
@cdedmonds I would say the key part is exposing as much control as possible to make things adaptable to a wide range of scenarios.

For example:
- for people identification/tracking/counting it may be enough to get the 2D output of the DNN and no 3D model fitting
- for interaction we do want a skeleton emphasizing smoothness and plausible poses (like we have now)
- for pose estimation (especially in multi-sensor/computer scenarios) we want a skeleton that may contain noise, missing limbs (with a confidence score) and less plausible poses at times but with no inter-frame smoothing at all
- for machine learning we probably want something in between, plausible poses but no inter-frame lag
- for offline tracking we may want to have a non-realtime version that favors accuracy over computation time

Again I don't know what is possible with the underlying algorithms and am just guessing, but the more that can be exposed the more wide the use cases. We're clever programmers using this you know :)

billpottle, 2019-07-17:
I definitely agree that if you can expose more control parameters, more possible use cases open up in a non-linear fashion, because then developers can work around the limitations by adapting their workflow, or by combining outputs from processing a stream more than once. I think this is particularly true for a device like the Kinect which feels like a revolutionary "general purpose" technology, with a lot of applications that haven't really been developed before. For instance, we would love to have low-latency, low computation positional accuracy. However, if that's not possible, we can always send the mkv files for remote processing on cloud GPUs and then send the data back to the end user.

Brekel, 2019-07-17:
Absolutely!
Although cloud is not always an option since body tracking deals with people and possible privacy issues that may require local compute. :)

rfilkov, 2019-07-18:
@cdedmonds Please also look at what the competition is doing: https://developer.apple.com/videos/play/wwdc2019/607

Chris45215, 2019-08-29:
For my use cases, the tracking latency and the ability to recognize a person are the key concerns - positional accuracy within 6 inches (roughly 15cm) is sufficient. Multiple sensors can be used to increase accuracy, but there is no way to get around the latency and tracking delay. 60fps depth tracking would probably help alleviate the problem, but if that's not possible with the hardware then reduced latency is a priority. We can always add more cameras, but we can't subtract time. Though I note it would be easier to add more cameras if they didn't each require a PC with a Nvidia card; Intel NUCs would take far less space.

rfilkov, 2019-09-01:
I must admit, v0.9.2 has significant improvements in means of performance and accuracy. In this regard, I have a question: do you change the model as well, or only the SDK internally?

hmexx, 2019-09-02:
@rfilkov Is there an MSI for 0.9.2? I can't seem to get my hands on it.

rfilkov, 2019-09-02:
@hmexx [here](https://www.microsoft.com/en-us/download/details.aspx?id=100128) it is.

qm13, 2019-09-03:
@rflikov v0.9.2 includes an updated model.

hmexx, 2019-09-06:
The new model does seem slightly better.

The main issue for us is that the model does not output confidence scores for the joints. Most 2D pose estimation have this feature, which makes it easy to deal with situations where the model has to fail (occluded joint). Is this something that can be added? I'll start a new issue with a feature request.

PierrePlantard, 2019-09-06:
I'm agree for the confidence score fro each joint. 
This improvement seems to be under evaluation (https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/38166871-joint-tracking-state).
Another big limitation is the hardware requirement. 
We want to work in mobility with laptop but such hardware (e.g. GTX 1070 or RTX 2070) need to much power to work in mobility (in battery mode).
We think it is very important to add more configuration possibilities for the body tracking. For example à lighter model compatible with lower grade graphics cards (GTX 1650 for example).

rfilkov, 2019-09-06:
@hmexx Here is a feature request in this regard: https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/38166871-joint-tracking-state Please upvote or comment there, too.

rfilkov, 2019-09-06:
@PierrePlantard Please upvote this feature request, as well: https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/38129473-body-tracking-without-cudnn

PierrePlantard, 2019-09-06:
> 
> 
> @PierrePlantard Please upvote this feature request, as well: https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/38129473-body-tracking-without-cudnn

Already done few weeks ago @rfilkov :)

vpenades, 2019-09-11:
Regarding pose smoothing, I never understood why low level APIs already deliver smoothed skeleton.

Skeleton smoothing is relatively inexpensive and easy to compute. Actually with kinect2 we developed our own skeleton smoothing algorythms based on our needs.

In other words, smoothing the skeleton is a post-process that can be done either with an additional call to a "SkeletonToolkitAPI" provided by the body tracking SDK itself, or by a third party library.

So there's no point in the output to be smoothed by default.

Brekel, 2019-09-11:
You can turn it off in the latest v0.92 release.
It proves the more settings/controls we have in the sensor/tracking SDK the better it can be adapted to different use cases.

Chris45215, 2019-09-11:
@PierrePlantard I've used the Azure Kinect for body tracking with a GTX
1060 and it works, and uses at most about 20% of the GPU according to
Windows Task Manager. I've not compared it's performance to a GTX 1070 or
greater, so I can only confirm that it does run and returns results. It is
noticably slower than the tracking of the Kinectv2, but that seems to be
true with even the GTX 1080.

The sensor is intended to track multiple people but my tests were done with
only a single person, so it is possible that the 1070 specification is a
high estimate to ensure it can track 6 or more people simultaneously.

JMonkey94, 2019-09-12:
Hi All, may i know the latest model version of azure body tracking sdk 0.9.2 solved the too slow and inaccurate issues? I can't fine any feedback or comment so far for this latest version.

PierrePlantard, 2019-09-12:
@Chris45215, Yes the body tracker also work in my laptop with a gtx 1050ti,
but only for 15 fps.
When I put my laptop to battery mode, the fps drop down to 5 fps due to
power throttling of the laptop.
These low fps induced a important lag, which is unacceptable for my use
case.
A good configuration feature (available for many deep learning models), is
the possibility to chose between accuracy and computing power required.

qm13, 2019-09-23:
We released 0.9.3 last Friday. It includes a brand new ML model (v2), improved 3D model fitting, smoothing off by defaults, and improved performance. Love to hear from the community on whether this release improves your apps.

hmexx, 2019-09-24:
Hey there.

we just tried your new model and it is definitely improved. Well done!
Unfortunately without joint confidence it’s still unusable for us. Even
with this improved model, when a joint is occluded, it will end up in a
random location. A low joint confidence would allow us to spot and deal
with such situations accordingly. For example by not visualising the joint
or by prompting the user to take some action.

RoseFlunder, 2019-09-24:
Same situation. I appreciate a better model of course but when fusing joint data from multiple kinects in a room I don't want to fuse this false joint data from joints that are occluded.

Chris45215, 2019-09-24:
qm13,
I compared V0.92 to V0.93 back-to-back with the bundled Body Tracking Viewer app, and the tracking latency (the time delay between reality and the output being shown on screen) for both versions is 0.25 seconds. This appears to be due almost entirely to a 0.2 second delay between the camera sensor (whether using RGB or depth) and transmission to the PC - it's as though the camera is buffering 5 or 6 frames internally, for no reason. For comparison, the Kinect For XBox One has a body tracking latency of around 0.12 seconds and half that for RGB and depth sensor data. For more modern comparisons, the Oculus Rift S's cameras have about a 0.02 second latency for room tracking, and some of the current Intel RealSense cameras boast a 0.006 second latency for similar tasks - that's not body tracking, but it shows how quickly the data can get from the camera sensor to a processed output. 

 
I made this simple test procedure anyone can use to evaluate performance, most notably the latency and lag:
1: With the Kinect plugged in, start the bundled Azure Kinect Body Tracking Viewer app. Maximize the output window.
2: stand in front of the Kinect while facing the computer screen, ensure you can clearly see the screen.
3: pull out your phone, start a camera or videocamera app, and set it to the highest framerate available.
4:  Tell your phone to begin recording, and point it at your computer screen so it can see the body tracking output. We'll assume you are holding the phone with your right hand.
5: Reach out with your left hand (or whichever hand isn't holding the phone), and ensure that you can clearly see your extended arm on the Body Tracking output, and can also see that hand on your phone's camera.
6: Suddenly and rapidly pull that extended hand downwards, while holding your phone camera steady.
7: repeat 5-6 a few times, then end the recording.

This gives you good, well-timed footage in which you can count the frames in your phone's video between your real movement and the Kinect's detection of that movement. My phone records at 60 fps, so there were 15 frames (0.25 seconds) between my movement and the Kinect's detection of the movement. 
I tried this again in the bundled Azure Kinect Viewer 1.2 with each the RGB and depth output, and those had delays of 0.20 seconds - better, but not by much. For a sanity check and as a 'control', I did the same test except filming the computer mouse and the cursor movement across the screen - that yielded a lag of about 1 frame at 60fps, or 0.016 seconds, so it confirmed that my monitor is sufficiently responsive and not the cause of the delay.

It seems that the problem may not be the tracking algorithm itself, but rather a very long processing time and delay for the camera data overall - assuming that the depth image output isn't being delayed to match the skeleton tracking. A delay that long is a non-starter for my application, as we aim for 0.05 seconds but can stretch a bit beyond 0.1 seconds if the tracking is perfect. 

I performed these tests on a PC with an AMD 2700X CPU and GTX 1060 GPU. Even if the GPU is below the recommendation, that shouldn't affect the 0.2 second latency between the camera sensors and the software output.

qm13, 2019-09-24:
@hmexx @RoseFlunder please see https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/734 re joint pose confidence.

@Chris45215 thank you for the excellent perf analysis. I have flagged your comment to the Sensor SDK team. You also may want to republish your analysis against the Sensor SDK.

rfilkov, 2019-09-24:
v0.9.3 is significant improvement for body index map accuracy. There are still some inaccurately classified pixels, in the area of the fingers and around the body edges, but well done so far.

Regarding performance: I don't see major performance boost in comparison with v0.9.2, but maybe I'm missing something or it's due to my Nvidia GTX 1060. Indeed, the suggestion of @Chris45215 is great for measuring the overall performance. It should be applied to the Azure Kinect Viewer, as well.

The suggestion for pose confidence levels is also a significant improvement on its own. I hope you can add the hand joints and hand states soon, too. It's already work in progress, as far as I see, and this gives hope for further improvement.

One cosmetic request: Could you please modify a bit the Body Tracking Viewer in the next release, to start full screen instead of in a small window.

billpottle, 2019-09-24:
@qm13 - 0.9.3 seems to have been improved the tracking for the arms, but to me it actually looks a bit worse for the legs. In 0.9.2 and before the standing leg was accurate while the extended leg was snapped back to the standing one - in 0.9.3 it seems to be 'splitting the difference' and both legs are now shown venturing into in the empty area between them. 

I have tested this both with pre-recorded MKV files in several directions and in the newly downloaded 3d viewer. 

Please see an example in issue [738](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/738) where I redid the analysis of a previously recorded file with 0.9.3.

YingzheLiu, 2019-09-26:
hey anyone knows how to record body tracking? Can we use k4arecorder to record it?

RoseFlunder, 2019-09-26:
No, the mkv file doesn't contain body tracking information but it has the color, depth, IR tracks. So if you open the recording with the playback API from the SDK you can feed the captures into the body tracking as usual as you would do it when working with a real device.

rfilkov, 2019-10-22:
@yijiew @qm13 Thank you for the latest update v0.9.4 featuring hand & finger joints tracking, as well as the joint tracking states! These are all very helpful. 

What I noticed recently is that the handtip joints track the fingers pretty well. There are some misses from time to time, but generally the fingers are well tracked. On the other side, the thumb joints don't track the thumbs so well, especially when the user rises his/her forearms. Is there any particular reason for this?

By the way, do you plan to add hand states (in means of open hand/closed hand) soon, as well?

qm13, 2019-10-29:
Closing as performance issue being tracked in #816

pratikgd, 2019-11-21:
Dear Support Team,
As we have  **kinect AZURE DK** bodytracking sensor but the performance of this sensor is low as compared to the **kinect V2** on CPU mode.
So we want the bodytracking sensor who’s working is similar with the kinect V2, Hence please advice and do the needful.

fractalfantasy, 2020-02-17:
I have to agree with @pratikgd - "CPU mode" is completely unusabel. Have a look at how slow it performs in the video below  (Core i7, all cores running at full tilt so it's not a multi-threading issue):
https://www.dropbox.com/s/j1cnjephldp95vr/Azure-Kinect-BodyTrackingv1.0-CPU-Test.MOV?dl=0

The Kinect v2 random forests technique was much faster and more reliable, why can't we just use this? Surely excluding users without Nvidia GPUs isn't good practice.

fractalfantasy, 2020-03-19:
i’ve created a feature request to address the latency, inaccuracy and slow speed of Azure Kinect body tracking by implementing the Kinect 2 ‘random forest’ model as an option.. pls upvote:

https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/39945454-legacy-body-tracking-like-kinect-v2

cc: @Chris45215  @PierrePlantard 
@vpenades @JMonkey94 @RoseFlunder @pratikgd

Jessica-Yeung, 2020-04-10:
What is the technical difference (AI features) between Azure Kinect and Kinect V2's body tracking except the Azure Kinect uses a DNN model to derive the skeleton, the Kinect v2 uses a ‘random forest’ model?

rfilkov, 2020-04-16:
After patiently waiting for months, I've come to this final request. https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/40188511-please-replace-the-body-tracking-sdk-team

qm13, 2020-04-17:
The Body Tracking SDK is entirely separate from the Sensor SDK. We recognize that there are performance issues in both SDKs. Using a single E2E experience whilst demonstrating E2E performance issue it makes it hard to isolate the bottle necks. Can we please use this issue to focus on body tracking performance - the time taken from enqueuing a capture to popping a result. This should take no more than ~30ms on the recommended Body Tracking minimum hardware for 30fps. We also realize this is a high hardware bar and are working on a lite DNN model.

fractalfantasy, 2020-04-20:
Happy to see this thread reopened, and that's great that you are working on a lite DNN model.  

Right now these are the main issues with body tracking performance (I can open up a separate ticket for each one but they all seem to fit this thread title):

1) FPS
2) Latency
3) Cross-platform compatibility
4) Resource hogging (GPU)
5) Accuracy (dynamic movement)
6) Power efficiency

Concerned that a lite DNN model will only solve half of these issues at best. Can you tell me how you plan on solving these?

Is there any way porting over a random forest **option** could hurt the development of the DNN model? 

I know I'm sounding like a broken record here but I think it would solve all of the above, and would let you guys to develop the DNN model in peace. It's also quite popular on the [feedback forum.](https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/39945454-legacy-body-tracking-like-kinect-v2)

billpottle, 2020-04-21:
@fractalfantasy - Thank you for breaking down the issues. Obviously, every use case is different. However, I would think accuracy would likely be the most important, because the others can generally be solved, if in sub-optimal ways. For instance, resource utilization can be solved by adding more resources. For accuracy, adding computing power only delivers incorrect results more quickly. Power efficiency only adds a slight cost to the end user - accuracy completely halts a project.

vpenades, 2020-04-21:
@billpottle You're mistaken on the accuracy/computing power ratio.

As stated by some users, high end computers with a lot of power are simply not an option for many business models, including ours.

Once you accept the only valid platform is a low end computer you have a choice between:
- High quality poses at low frame rate.
- Low quality poses at high frame rate.

And by "low quality" I mean no less than what the Kinect2 already delivers.

Now, why would someone choose "low quality" at high framerates?  Because for Natural User Interfaces, it is CRITICAL, to be able to pick fast movements or gestures, even if some limb is lost sometimes.  Because with a High Quality tracker at low framerate, you miss ALL the fast movements, which actually results in much worse overall quality.

Any decent human person tracker to be used for NUI should run at 30fps, regardless of the processing power available.

fractalfantasy, 2020-04-21:
@qm13 @wes-b specifically how would porting kinect 2 body tracking over as a legacy  OPTION hurt development of the DNN model?

dotslinker, 2020-04-22:
I beg your pardon in advance for the long and somewhat OT post. 
The current Body Tracking SDK is almost unusable on most of the devices where Kinect V2 was running smoothly (*). The result of this, especially in these Covid-19 times, is that we almost cannot use the new Kinect for Azure for all those applications home-based that are currently desperately needed to keep people active and in shape. 
Let me humbly suggest to the K4A development team to consider opening a development branch of the previous SDK for Kinect v2 to be used with the new Kinect for Azure, of course paying the price of a less accurate tracking. I would say that for most of the interactive applications was already fairly good (we are ahppily using Kinect v2 + old sdk in a rehab center for children).
Other solutions (i.e. based on Intel Realsense D425 or D415), much simpler and less effective from many points of view, are already in place, and if there will be no changes in the current Kinect for Azure SDK development plans regards the use of GPU-intesive DNN for the BT, maybe many will switch to those ones from the Kinect.

qm13, 2020-04-25:
@fractalfantasy @billpottle @vpenades @dotslinker I love the passion and the Azure Kinect team wants to address your issues. The team was recently reorganized in to the new AI Platform team within Azure. This and COVID-19 has somewhat impacted productivity. Please bear with us. We want to make it possible for you to bring your applications to life with Azure Kinect. Here are responses to your questions/points (hopefully I have not missed some).
Starting with @fractalfantasy excellent breakdown of the overall problem:
1. FPS - putting aside the compute question this is restricted by the Azure Kinect camera i.e. 30 fps.
2. Latency - this is a compute issue. What sort of latency do you think is acceptable from enqueuing the frame to popping the result?
3. Cross-platform - body tracking uses the ONNX Runtime. They announced at Ignite 2019 that they were adding Intel and AMD support using Direct ML (consequently Windows only). We are working with the Direct ML to maximize performance of body tracking as part of this effort.
4. Resource hogging GPU - this is also a compute issue. We agree we need a lighter algorithm but (as pointed out by some maybe not light enough). The Direct ML path will also open the door to VSPs i.e. Myriad X and its next generation. What is too much GPU?
5. Accuracy - this one is very important to the team. The team (and customers we have spoken) to believe the current body tracking is more robust i.e. less failures to track. The goal is also to be more accurate. Please provide examples were we are not as accurate.
6. Power efficiency - this also is a compute issue. In truth VSPs are the best path to addressing this. Silicon specifically designed to run AI models is more efficient than a GPU.
Regarding providing a random forest option we will investigate what it will take to port the model to Azure Kinect. Open sourcing is probably not possible as the community would need to train the model which requires labeled data.

cavisions, 2020-04-26:
@qm13 Regarding your request - I really hope this message finds you well.
> The goal is also to be more accurate. Please provide examples were we are not as accurate.

Please find several examples of very poor accuracy below. The video is taken with 6 years old boy 122cm tall and rather slim posture. 

### Problem #1 "Separated hands forward movement"

[image]
[VIDEO separated hands](https://youtu.be/6taoTKpU98M)

### Problem #2 "Joined hands movement"
[image]
[VIDEO joined hands](https://youtu.be/3f52_DSmLkM)

### Problem #3 "Legs to the sides movement"
[image]
[VIDEO legs](https://youtu.be/-a-YPs5BvXw)

The problems are less intensive when it comes to adults, but the accuracy while working with children at this age is unfortunately not good.

fractalfantasy, 2020-04-27:
@qm13 thanks for addressing these issues - I've answered your questions and notes in the list below - a lot of the developer expectations on these have already been expressed so I've referenced a few:

----

**1. FPS** -  30FPS is fine given the sensor limitations, but it's expected to run at 30FPS on an Intel CPU like it's predecessor:
- _"Any decent body tracker to be used for NUI should run at 30fps, regardless of the processing power available."_ - [quoted ](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/514#issuecomment-617289840)from @vpenades 

**2. LATENCY** - Current latency is a .25 seconds on a gaming rig - its expected that the Azure Kinect match or beat the latency of its predecessor (0.12s) and contemporaries (0.006s):
- _"the Kinect For XBox One has a body tracking latency of around 0.12 seconds and half that for RGB and depth sensor data. For more modern comparisons, the Oculus Rift S's cameras have about a 0.02 second latency for room tracking, and some of the current Intel RealSense cameras boast a 0.006 second latency for similar tasks"_ quote from @Chris45215 who wrote an [excellent post](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/514#issuecomment-534617419) breaking down the latency issue.

**3. CROSS-PLATFORM** - current compatibility is with Nvidia 1070 and up - expectation is for Azure Kinect body tracking to be run on any ~$1000 laptop made in the last 5 years like it's predecessor. 
- _"The Body Tracking use cuDNN and a powerful gamer PC with an NVidia wheras KInect 1 and 2 worked on simple core I3 / core i5 laptop years ago. Any chance to have a workaround with a lighter skeletal tracking ? It seems also the skeletal tracking is less accurate"_ quoted **Jean-Philippe Encausse** 
 one of the [highest ranked user requests](https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/38129473-body-tracking-without-cudnn) on the feature forum.
- _"The current Body Tracking SDK is almost unusable on most of the devices where Kinect V2 was running smoothly (*)."_ quoted @dotslinker

**4. RESOURCE HOGGING GPU** - To answer your question "What is too much GPU?" I say give us as much GPU as humanly possible. I'm pretty sure the Kinect v2 gave us 100% of the GPU, that would be perfect. 

**5. ACCURACY** - I have to disagree with you here, it isn't widely agreed upon that the Azure Kinect DNN Body Tracking is more accurate than Kinect 2. So far the[ only case](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1080#issuecomment-587421830) I've heard of it being more accurate is on static frames where a user is facing away from the camera, which is a pretty niche scenario. Accuracy has been reportedly much worse for dynamic movement than Kinect v2 and this accounts for most use cases of the Kinect.
- _" the DNN model performance is much worse than the body tracking of Kinect-v2. The joint positions are inaccurate by fast movements... I believe most Kinect users would expect better, more accurate and more performant body tracking experience, not worse."_ [quote ](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/514#issue-469244209)from @rfilkov who makes Unity's most downloaded Azure Kinect plugin.
- _"The current body tracking solution is too slow and inaccurate for any realtime uses."_ [quote ](https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/39945454-legacy-body-tracking-like-kinect-v2)from **Elie**
- _" Our entire project is basically on hold because Azure kinect body tracking is unusable for any dynamic motion."_ [quote ](https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/39945454-legacy-body-tracking-like-kinect-v2)from **William Pottle**

**6. POWER EFFICIENCY** - To answer your statement about VSPs - industrial clients shouldn't be expected to have specialized/experimental AI processors on hand, they will likely be using a high volume of cheap, small and efficient computer systems.

---

**TL-DR:** We expect Azure Kinect Body Tracking to beat his dad at everything, but right now his dad (Kinect 2 Body Tracking) is able to kick his ass on pretty much all fronts. Azure Kinect Body Tracking is still young and going though puberty and figuring out how to talk to girls, and one day he'll grow to be stronger than his dad, but right now we need his sturdy dad to help us build these houses.

All jokes aside, In order to make great application that will sell the Kinect we need the strongest body tracking option available now - not what will grow to be strong in 4 years. All your improvements have been very small and incremental so far, and will likely continue at this pace (especially in the midst of Covid). So we really need a random forest option to start building, and to finally leave you alone to make the DNN model exactly how you envision it.

dotslinker, 2020-04-27:
@qm13 Thanks a lot for your attention. I completely agree with @fractalfantasy. 

I beg your pardon if I'll be a bit long, but I'd like to tell you a story. 

Since 2015 I have been developing a C# app we are currently using in a pediatric rehabilitation lab with children. So far, we have been satisfied with Kinect v2 (and its SDK), included accuracy and fluidity (quite constant 30 fps on almost the PCs I tested it). Last January I went in Boston for the MIT xR Hackathon, where I tried to combine Kinect for Azure with Hololens 2 (ARehab team - https://vimeo.com/385983447). I was very much positive, since I found the Kv2 SDK very much easy and self-explanatory, and I thought the new version should have been the same or even better. 

Unfortunately, it was not the case. 

The "nightmare" started when I looked for the documentation, trying to figure out what to download in order to have all the proper tools up and running. Definitively it was not the same of the v2 SDK with its simple interface where you can easily select the feature you want to implement and in few minutes you were coding. 
Then I realized that also the source example was completely different, referring to some (undocumented) OpenGL functionalities that (on theory) required me to further foster that OpenGL world to fully handle the code. I was only in the need to grab the skeleton's joints positions and send them to the Hololens 2. I was using a DELL notebook equipped with an i9, 32 GB RAM and a GTX 1650 GPU, and I thought it was more than enough for the task. 

Unfortunately, it was not the case. 

It turned out that the demo C# source code, once compiled, ran on my laptop @ less than 25 fps (sometimes even much lower than 20 fps). The “footprint” of the whole binary folder was more than 600 MB vs my humble 16 MB of the size I was used to with my previous application (which included a lot of processing and charting algorithms too), with the two files for the DNN model and processing engine summing up the planetary size of around 500 MB...

Now, we do not need a spaceship to travel 100 yards: we need something to use now and in the near future, since most of the children that were coming to the lab for rehab activities cannot move anymore to our center. Hence we are currently looking at home environment to continue to delivery our treatments. The same happens also for other type of patients, especially elderly, who very likely will be forced at home much longer than the others. At the moment we cannot ask all these people to upgrade their PC to a newer one (paying more than 1000 dollars) just to use the K4A. Especially if we consider that the previous one (Kv2) was running smoothly on a very broad range of computers. 
Kv2 latency was ok, accuracy was enough for a “markerless frontal body tracking”, not only for gaming purposes but also for rehab activities. Kv2 source example were great as the documentation. From my point of view, the only two drawbacks of Kv2 are the cable connection (it was not a USB3 compliant connection), and… the current unavailability for purchase.
K4A appears to be a great piece of hardware, well designed and engineered. It is USB3 fully compliant (and this is great), and has many interesting features (like the tons of microphones). However, so far it seems too bulky to be largely used, mostly for the SDK, which seems to be still something for… hackers . We’d need just an extended version of the previous SDK to use the new Kinect. That will definitively pave the way to a universal adoption of the new device, which otherwise will be confined to research labs or technological enthusiasts. 
Please consider the opportunity you have (as Kinect for Azure developing team) to greatly help in these very tough times many people who now are forced (and will be likely for a while) at home. And more generally speaking to boldly contribute in the development of a novel way to treat people at home instead of making them travelling for hours just to get to the center where they may be treated.

PierrePlantard, 2020-04-27:
I'm totally agree with the previous comment. 
Accuracy during self occlusion, latency and hardware requirement limit the use of the Kinect for azure. 
I would nevertheless salute a notable improvement of the kinect for azure body tracking.
The ability to estimate a person's posture from the back. This is a very important feature for us and it will have to be accessible in the light version that is planned.

vpenades, 2020-04-27:
@qm13, Keep in mind many of us already have products leveraged against the Kinect2, and the new K4A body tracker drifts away too much from the Kinect2 in terms of features and performance to simply replace one by the other. In many cases it means a complete product redesign.

The mistake here has been to deliver the new DNN Body Tracking sdk BEFORE having a Kinect2 compatible SDK.

Over time, we could have _gradually_ adapted our software, and let clients that can afford it use the new DNN body tracking.... _without compromising existing clients_.

So, regardless of what other trained models and technologies are provided. Kinect4A needs a trained model _equivalent_ in features and performance to what the Kinect2 already delivered. This is what a lot of people in this forum has been asking all along.

Chris45215, 2020-04-27:
> Latency - this is a compute issue. What sort of latency do you think is acceptable from enqueuing the frame to popping the result?

I disagree that it is a compute issue, it appears based on my tests to be a bug. My data from processing a saved video showed a computation time of ~22ms per frame (RTX 2070 Super), so the latency between enqueing a frame and getting the result should be about 22ms. But my tests with live video showed that, even when running at a mere 5FPS, the system was returning results from 3 frames prior (with Microsoft's suggested code) or 2 frames prior (with my improvement on it). It sits on more frames when running at 30FPS.

Anyway, the entire thread is getting pretty bulky and unweildly. That's inevitable as it's not clear whether the body tracking latency is caused by delays in the camera, the depth SDK, or the body tracking SDK; but we should make new threads for separate issues. I would definitely say that the GPU requirement is worth its own threads - Microsoft made a major design error by requiring the user to have a powerful GPU rather than incorporating the CUDA cores and about 1GB RAM onto the camera, especially because any GPU with the requisite CUDA cores has at least 8GB RAM and other components that body tracking doesn't use, and thus the discrete GPU costs far more than it would cost for the needed components to be integrated to the camera. But that's an issue for CameraV2 to (hopefully) solve, if Microsoft decides to continue the product.

billpottle, 2020-04-27:
As far as accuracy, our main issue is when legs are higher than the waist or moving quickly, ie, with kicks. This has been a separate thread before #(738) You can see an example here: https://www.my-ai-coach.com/t-c5-1586983138 Go to the wireframe below and adjust it - it looks very little like the video.  In contrast, you can see much higher accuracy with slow moving hand techniques. https://www.my-ai-coach.com/t-c13-1586982826

fractalfantasy, 2020-04-27:
@qm13 @wes-b  I can assure you that none of us want to be here on github trying to convince you guys to fix the sdk, we'd all prefer to be coding 🤓

Chris45215, 2020-04-27:
@dotslinker in your case, if you want to use something different from the Kinectv2 (which is a perfectly good sensor), I would suggest looking at the Orbbec Astra, or Intel's Realsense (which will require Nuitrack's software for body tracking). These are both low-footprint options in terms of code size, physical size, and computer requirements. The only caveat is that Orbbec's software can be a bit janky if your try to compile the source itself, as I believe the last time I tried it I discovered that Visual Studio was referencing file locations that existed only on their developer's computers (there is probably some easy solution for this, I've just not needed to go through it). I have been very impressed by Orbbec's offerings; and while their code isn't perfectly compatible with the Kinectv2 or K4A SDK libraries, it returns results that are very similar and thus it is relatively easy to adapt your code to accommodate Orbbec's data format. These sensors also have the benefit of being cheaper, and powered entirely by the USB. Also, Orbbec has a stereo camera version of its sensor which - while it does not perform tracking quite as well as the ToF version - multiple stereo sensors can be used without mutual interference. Though I've not programmed multiple Orbbec sensors to simultaneously work with one program, so I can't promise that there won't be some kind of resource conflict.

K4A's advantages are that it has the most reliable and accurate body tracking, even if it does have high latency, and that it most easily allows multiple sensors to work together without interference. The later is somewhat negated by the requirement of the GPU, as the most powerful GPUs on the market can only handle body tracking from 2 or at most 3 sensors simultaneously. But, it's a feature.

NickAtMixxus, 2020-04-27:
Agree on many of the recent points that have been addressed here by @fractalfantasy @vpenades @dotslinker and more.
@qm13: Thanks for detailed update. Regarding "What sort of latency do you think is acceptable from enqueuing the frame to popping the result?"
I can only speak for my needs but for MixxusStudio interactive exercise, rehab apps/games I'd say the Kinect V2 latency is quite OK, it would not hurt if it was slightly better. In my experience the precision is also fine if the whole body is completely in its frustum, "visual field".

For interactive apps I would compare the need for short latency similar to playing virtual instruments.
The real world is not instant, there is a slight inherant "latency" in an objects weight or inertia. 
As a comparison, if we want to experience that we are naturally playing an instrument the roundtrip latency needs to be around 20 ms or less, for some up to around 30 is acceptable but not much more, above that it starts to feel unnatural, most won't like to go above 10-20 ms.
That is from pressing the keyboard to hearing the sound.
To get a natural visual feeling of moving a virtual hand or interact with virtual objects the latency ideally should be somewhere there.
(Our brains are somewhat slower to react to visual stimuli than to audio so it may make us more sensitive to delayed visual stimuli but I'm speculating now.)

What the Kinect for Azure is capable of regarding latency I don't know but if you could make it run like the Kinect V2 then please make such an option.
I thought the Random Forest idea sounded interesting, give us Kinect V2 without the bulky cables. A kind of Easy Mode option so it can be available for a lot of people (without heavy systems) that really can make use of it. And if that would be fairly easy to implement perhaps we could get is soon? If I may add, I really wish we could get a fast Kinect soon cause it's hard to continue developing when K2 is not really an option anymore but you don't know what will become of K4.

qm13, 2020-04-28:
Thanks for all the responses.
@cavisions yes, body tracking does not do well with kids under 8 years. This is primarily to do with body portions which do not become normal until around 8 years old (this according to a Dr we consult with). We are data limited. It is very hard to get captures of 2 - 7 year old kids.

@fractalfantasy whilst I agree regarding the K4W latency (being about 1/2 of AK) there other two you list are not body tracking uses cases. They are SLAM uses cases where you are predominantly reading an IMU and using vision to correct for IMU drift.
K4W required GPU. The Xbox One reserved >10% of the GPU to run Kinect even if the game did not use Kinect. This was the heart of the PS4 vs One debate that raged after the game consoles launched. That said I get that you want 80+% of the GPU for graphics.

@dotslinker thanks for the excellent story behind your use case. We are aware of a number of companies developing in home rehab systems that are seeing an uptick in interest in the last couple of months. To my VSP point these companies are developing take home hardware that plugs into the TV. The expectation is that patients will be loaned hardware for the duration of rehab.

@PierrePlantard yes, you should be able to track a person from the back with the lite model.

@vpenades sorry but there are no plans to provide backwards compatibility for K4W APIs. Assuming we do add support for random forest it will be through the current AK APIs.

@billpottle we have added high kick (and a few other extreme positions) to the training and testing data sets. Thank you for your contribution here. The high speed motion is more problematic and is not a body tracking issue per se. The Azure Kinect camera exposure time is longer than the K4W resulting in more motion blur. This was a HW design trade off that we hope for fix in the next generation of sensors

@Chris45215 there is a separate GitHub issue tracking device and Sensor SDK latency here #816 
We realize that there are alternatives to Azure Kinect and SDKs. If Azure Kinect currently does not work for you then you should absolutely look at alternative technologies. The Azure Kinect team is only at the start of its journey to democratize depth cameras.

[42 more replies]

closed 2020-06-27 by rfilkov