microsoft/Azure-Kinect-Sensor-SDK #517: high cpu utilization.

barakooda, 2019-07-19:
60% + cpu utilization when using kinect azure.

cdedmonds, 2019-07-19:
@barakooda there will be higher CPU utilization in the viewer than when using the SDK. Also note that at higher RGB resolutions more CPU is used to convert from the camera's native pixel format to BGRA so use the resolution that best matches your needs.

cdedmonds, 2019-07-19:
@barakooda I determined that the issue on my machine was unrelated. Can you provide some more details on SDK version, operating system version, and camera mode?

barakooda, 2019-07-20:
sdk version : 1.1
screenshots:
https://ufile.io/e2kj2ks5

bibigone, 2019-07-21:
Can it be related to MJPEG -> BGRA decoding? I found that this decoding in Sensor SDK is very slow and consumes a lot of CPU.

Gnimuc, 2019-07-22:
> 10-30% + gpu utilization...

This is very likely due to IMGUI on which the Azure Kinect Viewer is based. Merely running [the IMGUI demo code](https://github.com/ocornut/imgui#demo-binaries) will cost ~20% GPU utilization, even if no UI widget is being drawed on the screen.  But I don't think this is a big deal though, unused resources are wasted resources :).

wes-b, 2019-07-22:
Thanks, this usage is by design. We do expect an added CPU and GPU load with using the K4aViewer tool.

closed 2019-07-22 by wes-b