microsoft/Azure-Kinect-Sensor-SDK #519: Application crashes on attempt to create the second instance of k4abt_tracker_t

bibigone, 2019-07-21:
If application is trying to create two instances of `k4abt_tracker_t` simultaneously (that is, the second instance is created before closing the first one), application silently crashes.

Brekel, 2019-07-21:
Ran into the same issue.
I believe I read somewhere that the body tracker doesn't support multiple instances, would be nice if it can support that in the future (for multi sensor scenarios) and at least for now gracefully fail with a return value and log item.

wes-b, 2019-07-22:
We do only support 1 instance at a time. We need to update documentation and attempt to generate an error.

venscn, 2019-07-23:
Yes, there is still a problem. Shutdown and release can't release memory correctly.

yijiew, 2019-07-23:
@venscn Thanks for spotting the bug. I have a fix for the issue that shutdown/release can't fully release memory. You should see the fix in the next release. As for the issue @bibigone points out, it is by design. We don't allow user to create two instance of k4abt_tracker_t simultaneously. We are working on a fix that we will fail gracefully and log the error. 

BTW, @bibigone and @Brekel , why do you want to create two k4abt_tracker_t simultaneously? What is the use case for this? If the use case is valid and important, we might consider adding this feature.

bibigone, 2019-07-23:
> BTW, @bibigone and @Brekel , why do you want to create two k4abt_tracker_t simultaneously?

For multi-sensor configuration, for example. Skeletal data in conjunction with point clouds can be used to determine mutual sensors position.

Brekel, 2019-07-23:
> BTW, @bibigone and @Brekel , why do you want to create two k4abt_tracker_t simultaneously? What is the use case for this? If the use case is valid and important, we might consider adding this feature.

@yijiew
Indeed, I'm also running multiple sensors on one machine (with a beefy GPU), each running in their own thread from a single application. So my thought was to run one tracker per sensor to keep things separate.

diablodale, 2019-07-24:
Since Kinect v1, I've created Kinect software for artists, researchers, and educators. These groups often provide the flashy and exciting demos, videos, and "buzz" articles that generate excitement and press. I have *repeatedly* been asked by my customers over all those years how they can use two sensors on a single PC to track humans in their solutions.

* v1 new concept to everyone and USB2 days with computers not yet able to really process/render all the data.
* v2 hungry customers wanted multiple sensors. MSFT devs in the v2 alpha started to implement multiple sensors but gave up giving a key reason of the rarity of two USB3 controllers in the same computer. I have the v2 alpha code that would have used these multi-sensor APIs.
* K4A? Microsoft, devs like those on this thread, and all our customers all know this problem area well. My customers are still hungry. I had a customer *this week* ask me how to use multiple sensors for tracking with the v2 Kinect.

I want to tell my customers "a solution is coming...you can buy two K4A sensors and use my software". But, if the Microsoft team doesn't provide multiple sensor support, then the K4A story is only an small increment better than the already solid Kinect v2 which has well tested code and a solid aftermarket.

bibigone, 2019-07-25:
@yijiew

> I have a fix for the issue that shutdown/release can't fully release memory. You should see the fix in the next release.

Does it fix issue#541 also?

cdedmonds, 2019-07-25:
@bibigone Yes that should fix #541. For this issue we are tracking a fix that would specifically give an error message (instead of crashing) when more than one body tracker instance is created in the same process.

Chris45215, 2019-07-28:
> BTW, @bibigone and @Brekel , why do you want to create two k4abt_tracker_t simultaneously? What is the use case for this? If the use case is valid and important, we might consider adding this feature.

I have a use case that I can't elaborate in detail on an open forum as some patents are still being filed. Multiple cameras are needed because redundancy is mandatory for the application, and because the tracked person's body may be occluded by other objects. Those other objects can be larger than the tracked person, so the only way to determine the person's location is through multiple cameras. The Azure Kinect is the cheapest part of the system (by an order of magnitude), so we expect the smallest installations to use a minimum of 3 Kinect's, and each customer will have multiple installations.

The market for this system is enterprise customers, and it is the exact same enterprise customers that Microsoft is targeting for Azure IoT and similar services, so this is an excellent way to pull them into the Azure ecosystem. I am surprised that Microsoft has not presented this particular application as the leading use-case for the technology.

closed 2019-08-06 by cdedmonds

ZhengzhongSun, 2019-10-05:
Does body tracking sdk support multi kinects now? @cdedmonds

richardeakin, 2020-03-03:
As of [0.9.5](https://docs.microsoft.com/en-us/azure/kinect-dk/body-sdk-download#v095), body tracking with multiple devices is listed as supported, however in my tests a third device kinected will not track bodies, and there is no error message. Is this a known issue?

yijiew, 2020-03-03:
@richardeakin Did you pass frames from different sensors to the same body tracker? For multiple device to work, you will need to create different trackers for different devices.