microsoft/Azure-Kinect-Sensor-SDK #521: Skeleton smoothing is too aggressive for low FPS

bibigone, 2019-07-21:
It looks like that Body Tracking SDK applies some kind of smoothing to skeletal tracking results. And this smoothing is too aggressive, especially for frame rates lower than 30 FPS (15 FPS, not mentioning 5 FPS). This results in significant inaccuracy: skeleton is behind real pose in terms of frame-to-frame movement:
[image]

Is it possible to add some setting for this smoothing?

Brekel, 2019-07-21:
I noticed the same thing and fully agree, it seems it's even over smoothing at 30 fps. (I didn't try 15/5 fps actually)
Here's a (quite popular) feedback ticket I filed a little while ago that you can up-vote: https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/38027029-user-adjustable-skeleton-smoothing

tesych, 2019-07-22:
duplicate of #514

closed 2019-07-22 by tesych