microsoft/Azure-Kinect-Sensor-SDK #526: Is there API supporting to calculate color image uv's corresponding uv in depthimage

Ayangdanny, 2019-07-23:
k4a_calibration_2d_to_2d can do this thing ,but it need source_depth_mm as an input, but I dont know the depth value in colorimage. 
I need use k4a_calibration_2d_to_3d calculte depth in colorimage and use k4a_calibration_2d_to_2d  to find its corrsponding uv in depthimage?

Ayangdanny, 2019-07-23:
Is there API like realsense which use a search strategy to calculate the color's corresponding uv in depthimage.
https://github.com/Ayangdanny/librealsense/blob/master/include/librealsense2/rsutil.h

venscn, 2019-07-23:
@Ayangdanny 
Transformation_depth_to_color_camera

Ayangdanny, 2019-07-23:
@venscn You did not understand what I mean. This API trans depthimage to color, not from color to depth and also the output is a transformed image, I still can not get the one by one uv corresponfdence between color and depth image .

venscn, 2019-07-23:
@Ayangdanny
U can see issues #260 or #98.
I don't know if it is the same as your question.

Ayangdanny, 2019-07-23:
@venscn  they are not.

rfilkov, 2019-07-23:
@Ayangdanny Isn't it similar to this issue: [#502] ?

Ayangdanny, 2019-07-23:
yes，but #502 map the whole frame，it does not care the single pixel.  For example,  one pixel uv(300, 200) in color image. I want get the uv of the correspondence pixel in depthimage.
you can see the function(rs2_project_color_pixel_to_depth_pixel) of realsense , it is what I want.
https://github.com/Ayangdanny/librealsense/blob/master/include/librealsense2/rsutil.h

venscn, 2019-07-23:
U don't need depth info to mapping the color image?

venscn, 2019-07-23:
@Ayangdanny
Hoho, u want to point to point.

venscn, 2019-07-23:
@Ayangdanny
U can make depth equal 1, and .... lookagain  tomorrow, sleeping now.

rabbitdaxi, 2019-07-23:
@Ayangdanny , please take a look #498 and the updated documentation (https://docs.microsoft.com/en-us/azure/Kinect-dk/use-calibration-functions#converting-between-2d-coordinate-systems), this line particularly: "The function requires the pixel’s depth value (in millimeters) in the source camera image as an input, one way to derive the depth value in the color camera geometry is to use the function k4a_transformation_depth_image_to_color_camera(). "

mbleyer, 2019-07-23:
@Ayangdanny, you are right that we do not support the exact function that you are looking for. In general, we suggest using k4a_transformation_depth_image_to_color_camera() to obtain the depth value in the color camera system.

However, if you want to avoid transforming the whole depth image, the following search strategy might work. You can iterate over all discrete depth values d. For every depth value d, compute uv' = k4a_calibration_2d_to_2d(uv, d, Color_to_Depth). Look up the depth d' of pixel uv' from the depth map. Now compute uv'' = k4a_calibration_2d_to_2d(uv', d', Depth_to_Color). If uv'' equals uv, uv' is the corresponding pixel you have been looking for. Please note that such a roundtrip will not always be possible, as the corresponding pixel in the depth map might be occluded.

Ayangdanny, 2019-07-24:
@rabbitdaxi @mbleyer  Thanks, I know the search strategy,  I suggest you add this function later. This feature is often used and can avoid  performance waste when just need a few pixels.

PR #696 (2019-08-30): Add API to transform pixel from color image to depth image with searching on epipolar line

## Fixes #526 

### Description of the changes:
- API to transform pixel from color image to depth image without requiring transform the entire depth image to color camera for getting the corresponding depth, instead, we search on epipolar line to find the best candidate
- Unit test

### Before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [x] I [built my changes](https://github.com/Microsoft/Azure-Kinec [... cut]
closed 2019-08-30 by rabbitdaxi