microsoft/Azure-Kinect-Sensor-SDK #530: Multiple Device Sync Issues

wes-b, 2019-07-23:
Multiple Device Synchronization has a couple issues that prevent it from working reliably.

**Issues that the user must be aware of and work around**
- [x] #471 Color to Depth Timing may have an offset if setting manual exposure repeatedly.

**Issues that are actively being worked on / fixed
- [x] #362 [***TARGET SDK VERSION is 1.2.0***] Setting exposure does not result in the expected exposure value
- [ ] #448 Power Line setting is being changed by firmware after being set by SDK
- [ ] #529 Subordinate timing is off by ~100us

YuXinFan, 2019-07-26:
I meet some problem with sync. When connects external sync only one master which is not a ring:
1. In some running case, the timestamp of master capture is 100,000 usec forward. In this case, the other devices are aligned .
2. In some running case, the timestamp of some devices is about 30,000 usec lag, but they sometimes can also have the same timestamp in a loop round.
I test many times. And I want to get stable sync. Why this problem occurs?

wes-b, 2019-07-26:
I have not heard of the timestamp being that far off, but I am able to run 3 devices at my desk at 30FPS (no image processing), where I have seen it off by 4-5 ms. 

The problem is from #362, you need to be setting a manual exposure value. But what you set / get via the API is different then what is actually used and returned via k4a_image_get_exposure_usec. We also have a firmware fix coming. 

For now try building a version of the SDK with the fix for #362 (until 1.2.x is released). The timing won't be perfect yet, but is should get you within a few milliseconds or less.

initialneil, 2019-08-12:
I'm having the same sync issue. I'm connecting 3 devices to one PC, wire-synced one after one. One of the device is master and the others are subordinates.
When I use system timestamp from color image, the results look synced, yet the device timestamp are far from close to each other.
See the screenshot below. The device timestamp was divided by 1,000 and system timestamp divided by 1,000,000:
[image]

closed 2020-04-20 by wes-b

initialneil, 2020-07-20:
Hi, I just want to give an update on my previous problem. I've only recently realized that ALL subordinate devices must be started before the master. After I fix this, my sync problem seems to be gone, the device timestamp works now.
My bad. But I suggest in the white paper of sync, make sure people notice this. Thanks.