microsoft/Azure-Kinect-Sensor-SDK #532: how to use the simple viewer example

wqs111000, 2019-07-24:
How to use this [ simple viewer example](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/viewer)? Could anyone add some detailed guidance? 
I built this example by cmake (Win 10, VS2017, x64) and opened the  Project.sln. How to set its properties? The additional dependencies are weird.
**Screenshots**
[image]

I tried to change some properties in VS++ directories and Linker, and added the head files of imgui library. But I was stuck by the errors below. How to fixed them? Thanks.
**Screenshots**
[image]
[image]

cdedmonds, 2019-07-24:
@wqs111000 Did you use Visual Studio to open the CMake project or try to create a new solution?

williampMSFT, 2019-07-29:
This looks like the linker can't find imgui.lib - I notice that you have a warning hidden in your error view, what is that warning?  IIRC if it can't find a lib, it treats that as a warning and then errors if you actually call stuff from that lib, so I suspect that the warning says 'couldn't find imgui.lib' or similar.

How did you generate this MSBuild solution (i.e. exactly what command did you use)?

wqs111000, 2019-08-02:
Thanks a lot  for your replies.
>Yes, an error said  "couldn't find imgui.lib" once occurred. But as far as I know,"Dear ImGui is self-contained within a few files that you can easily copy and compile into your application/engine ". So I removed the imgui.lib  in the additional dependencies and copied related files of ImGui to the project path. But, it seemed the imgui did not work.

>I used the cmake GUI to generate  this project  by  setting source and target paths and other default settings for visual studio 2017.

williampMSFT, 2019-08-02:
Ah, I see.  When you removed imgui.lib from the additional dependencies, you stopped "compil[ing ImGui] into your application/engine".  That means that all the things that are calling it can't find it anymore.

I suppose you could directly compile all the files that are currently getting linked into that lib into some other lib that you link in, but one way or another, you need to compile and link in ImGui or you're going to get those errors.

Either way, the behavior you're seeing is the behavior one would expect if one removed a library that one was calling, which it sounds like you did, so I'm going to close this issue.  I'd recommend deleting your CMake cache and regenerating it to get back to a working state.

closed 2019-08-02 by williampMSFT

wqs111000, 2019-08-05:
I saw the problem. The related .cpp files of imgui  were  not all added to the project. After  the files added, the project could work.
[image]