microsoft/Azure-Kinect-Sensor-SDK #533: Solution to TOF "occlusion noise"

barakooda, 2019-07-24:
either from one or multi camera data, 
if you guys can offer state of  the art 
realtime TOF "occlusion noise" reduction post process.
it will be great.
(currently i only delete points by distance or triangle size)

[https://www.youtube.com/watch?v=0FBY_wlY9pQ](url)

keep up the great work

cdedmonds, 2019-07-24:
@barakooda this may be an area that we could publish samples. Different applications will likely have slightly different filtering needs.

mbleyer, 2019-07-25:
@barakooda, can you specify what exactly you mean by _occlusion_ noise?

barakooda, 2019-07-26:
Well I`m not using scientific terms :-) or expert on this side.
but,as far as noticed,
there is "regular" noise when you for example point the sensor onto a wall and sees running bumps.
and 
there are different kind of noise \ artifacts behaviors on the edges...
I can understand maybe its interpolation outcome between gaps on the Z .
but it can really behave weird,maybe unpredictable in some cases... (?)
so i called it occlusion noise :-)

cdedmonds, 2019-08-07:
Currently this is outside of the scope of the changes we are considering for the sensor SDK.

closed 2019-08-07 by cdedmonds