microsoft/Azure-Kinect-Sensor-SDK #534: The best TSDF \ other realtime mesh solution microsoft can bring.

barakooda, 2019-07-24:
Yeah - "new" buzz term - "volumetric content"
I will happy to see it as fundamental brick in the evolving kinect sdk. 
https://computergraphics.stackexchange.com/questions/8068/generate-mesh-from-pointcloud-data

Keep up the great work.

cdedmonds, 2019-07-24:
@barakooda please take a look at the sample here and let me know if it is what you are looking for:
https://github.com/microsoft/Azure-Kinect-Samples/tree/master/opencv-kinfu-samples

rabbitdaxi, 2019-07-29:
@barakooda , please let us know whether above answer addressed your issue.

barakooda, 2019-07-29:
Hi sorry,
I cant answer this question in a serious way,
I will have to benchmark my self.

As far as i noticed kinfu offer inferior performance and quality over google team solution and 
one other company i take part in.

also I would recommend to add  higher quality offline mesh creation option 
such as Poisson Mesh.

Hopefully will able to answer this question in the next two months after benchmark and creating
full pipeline my self .

closed 2019-07-30 by tesych