microsoft/Azure-Kinect-Sensor-SDK #535: The depth data of eyeball mouth and nose on a human face is of poor precision

Ayangdanny, 2019-07-24:
I use the pointcloud data generated a human face mesh.  but the eye mouth and nose  have poor  precision as you can see from the attached files. There are two holes in the eyes. the lip edge is too sharp which is different from the true data.  The third picture is more accurate which is captured by other RGBD device. Actually we found Kinect One also have the same problem before. 

camera configuration:
k4a_device_configuration_t config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
config.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;
config.camera_fps = K4A_FRAMES_PER_SECOND_30;
config.color_format = K4A_IMAGE_FORMAT_COLOR_BGRA32;
config.color_resolution = K4A_COLOR_RESOLUTION_720P;

the distance between camera and face is about 700mm.

[image]
[image]
[image]

cdedmonds, 2019-07-24:
Thanks for the report. Do the lip edge and nose edge look different if the camera is placed lower relative to the face?

cdedmonds, 2019-07-30:
@Ayangdanny thanks for reporting this issue. I believe the issues you are seeing are artifacts of the camera hardware and not software. For now I am resolving this bug as by design. Please reactivate the issue if there are specific changes you would like to see.

closed 2019-07-30 by cdedmonds