microsoft/Azure-Kinect-Sensor-SDK #536: How to use the microphone

nicolascarre, 2019-07-24:
Hello,

Is there a way to get the sound received from the microphone ?
Actually I just need to know what is the volume in decibel.
From some others device, I was retrieving a sample array, then compute the volume in db.
I see nothing about the microphone in the sdk.

Thanks for your help

cdedmonds, 2019-07-24:
The microphone is a available through platform standard recording APIs on Linux and Windows. If you are looking for integration with Azure Speech Services look here:

https://docs.microsoft.com/en-us/azure/cognitive-services/speech-service/

nicolascarre, 2019-07-24:
I have already researched about Speech, I don't find anything about a sample array.

xthexder, 2019-07-24:
The Sensor SDK does not currently support accessing the microphones through the API. The system-level audio APIs should be used instead.
The Kinect Viewer app uses libsoundio, which is a cross-platform audio library. You could also use a platform-specific api like WASAPI on Windows, or PulseAudio on linux.

Here is some relevant code from the viewer that correlates microphones to a specific kinect device: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/tools/k4aviewer/platform/windows/k4adevicecorrelator.cpp

cdedmonds, 2019-07-25:
This is an area we could improve the documentation.

nicolascarre, 2019-08-07:
I can't compile the k4aviewer from the source code, I have already tried a lot of things, nothing is working.
Is there somewhere a good documentation of how to integrate the libsoundio ?

wes-b, 2019-08-07:
> I can't compile the k4aviewer from the source code, I have already tried a lot of things, nothing is working.
> Is there somewhere a good documentation of how to integrate the libsoundio ?

Hi @nicolascarre please open a new issue and follow our bug template.

nicolascarre, 2019-08-07:
Here is the new issue :
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/606

PierrePlantard, 2019-12-11:
Hello,
Any update expected about using the microphone ?

tesych, 2020-01-22:
@PierrePlantard, the Azure Kinect DK microphone is supported by [Speech SDK](https://docs.microsoft.com/en-us/azure/cognitive-services/speech-service/speech-sdk).  Please follow the link to see the full documentation.

closed 2020-01-22 by tesych

xiaotaw, 2020-04-14:
@nicolascarre @PierrePlantard --- I tried a third party library odas for audio recording: https://github.com/xiaotaw/odas.git. Hope it helps.

PierrePlantard, 2020-04-14:
Hi @xiaotaw,
thank you for the link, we will test it !

ttxiaoxitt, 2020-04-28:
> Hello,
> Is there a way to get the sound received from the microphone ?
> Actually I just need to know what is the volume in decibel.
> From some others device, I was retrieving a sample array, then compute the volume in db.
> I see nothing about the microphone in the sdk.
> Thanks for your help
Hi, nicolascarre.Do you resolve the problem? I had a same question, too.

nicolascarre, 2020-04-28:
Hi @ttxiaoxitt,
Yes we finally found a solution. We use the NAudio library.
If you need help I can provide you a sample.

keenanfitzharris, 2020-07-13:
Hello @nicolascarre !

I am in a similar situation as you currently. I am trying to access the microphone array data retrieved by the Kinect DK (in decibels of each microphone). I have looked into using ODAS or WASAPI. Do you mind helping me? It appears you were able to arrive at a solution.

yuchienhsiao, 2020-09-23:
Hi, @nicolascarre 
I would like heve the sample, please
thank you very much.

isaacwong96, 2020-10-13:
> Hi @ttxiaoxitt,
> Yes we finally found a solution. We use the NAudio library.
> If you need help I can provide you a sample.

I'm in a similar situation,too.May I have the sample?
Thank you very much.

jessekirbs, 2021-05-18:
> 
> 
> Hi @ttxiaoxitt,
> Yes we finally found a solution. We use the NAudio library.
> If you need help I can provide you a sample.

@nicolascarre Hey Nicolas - would you please provide your sample using NAudio to capture audio? It would be most appreciated. Thank you!

sanweiliti, 2021-08-19:
> Hi @ttxiaoxitt,
> Yes we finally found a solution. We use the NAudio library.
> If you need help I can provide you a sample.

@nicolascarre Hi Nicolas, is it possible for you to share the NAudio audio capture sample? Thanks a lot!

knat, 2021-08-19:
Try Microsoft Media Foundation: https://docs.microsoft.com/en-us/windows/win32/medfound/audio-video-capture-in-media-foundation

TristanWYL, 2021-09-08:
I believe this [repository](https://github.com/TristanWYL/Azure-Kinect-Recorder) could help! With it you could know how to access to the raw data of the microphone array, and how to save the color image with C#.

busybeaver42, 2024-01-22:
for C++ have a look: 
https://github.com/busybeaver42/kv3
This example comes with the right cfg files for the kinect azure micro phone array.