microsoft/Azure-Kinect-Sensor-SDK #541: Crash when creating body tracker

hvallee, 2019-07-25:
Hello, we have an issue with the body creator. Each 5 or 10 times we launch it (after launching everything with the device, sensor, and closing everything as well, we experience a crash when creating the body tracker).
I saw in this issue: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/519
that there is a problem whith releasing memory. Could it be the same problem, and if so, do you know where the fix will be release, and if we can do something about it waiting for it?

Thank you.

**To Reproduce**
1: Loop on cycle of starting device, sensor, body tracking, closing everything.

The function used to create the body tracker:
///body tracking
bool KinectAzureDeviceManager::CreateBodyTracker()
```
{
	k4a_result_t TrackerResult = k4abt_tracker_create(&g_SensorCalibration, &g_BodyTracker);
	if (TrackerResult == K4A_RESULT_FAILED)
	{
		return false;
	}
	else
	{
		return true;
	}
	return false;
}
```

**Logs**
[image]

GPU Cuda raises to 100% when crashing
**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS with Version:  Windows 10
 - SDK Version:  1.0.1

bibigone, 2019-07-25:
I have the same issue. Indeed, most likely it is connected to the memory leaks discussed in issue#519.

cdedmonds, 2019-07-25:
Thanks for reporting. There is a memory leak in the body tracker that we are looking to fix in the next preview release.

closed 2019-08-06 by cdedmonds

nachoreimat, 2021-07-15:
I have the same proble. With version 1.0.1 everything works.
But with version 1.1.0, unless I use K4ABT_TRACKER_PROCESSING_MODE_CPU, for all the GPU modes I got this errors on the call `k4abt_tracker_create`

```
2021-07-15 17:29:05.6079611 [E:onnxruntime:, inference_session.cc:1268 onnxruntime::InferenceSession::Initialize::<lambda_d0b43ece64526f87430c2f161129335c>::operator ()] Exception during initialization: C:\source\ort-restore\onnxruntime\onnxruntime\core\providers\dml\DmlExecutionProvider\src\MLOperatorAuthorImpl.cpp(1487)\onnxruntime.dll!00007FFB47C3F2AC: (caller: 00007FFB47C8EE66) Exception(3) tid(2008) 80070057 The parameter is incorrect.

Exception during initialization: C:\source\ort-restore\onnxruntime\onnxruntime\core\providers\dml\DmlExecutionProvider\src\MLOperatorAuthorImpl.cpp(1487)\onnxruntime.dll!00007FFB47C3F2AC: (caller: 00007FFB47C8EE66) Exception(3) tid(2008) 80070057 The parameter is incorrect.

[2021-07-15 18:29:05.639] [error] [t=8200] [K4ABT] ..\src\TrackerHost\TrackerHost.cpp (185): Create(). Exception during initialization: C:\source\ort-restore\onnxruntime\onnxruntime\core\providers\dml\DmlExecutionProvider\src\MLOperatorAuthorImpl.cpp(1487)\onnxruntime.dll!00007FFB47C3F2AC: (caller: 00007FFB47C8EE66) Exception(3) tid(2008) 80070057 The parameter is incorrect.

[2021-07-15 18:29:05.639] [error] [t=8200] [K4ABT] ..\src\sdk\k4abt.cpp (38): tracker->Create(sensor_calibration, config) returned failure in k4abt_tracker_create()
cwipc_kinect: Warning: Body tracker initialization failed
2021-07-15 17:29:06.0254280 [E:onnxruntime:, inference_session.cc:1268 onnxruntime::InferenceSession::Initialize::<lambda_d0b43ece64526f87430c2f161129335c>::operator ()] Exception during initialization: C:\source\ort-restore\onnxruntime\onnxruntime\core\providers\dml\DmlExecutionProvider\src\MLOperatorAuthorImpl.cpp(1487)\onnxruntime.dll!00007FFB47C3F2AC: (caller: 00007FFB47C8EE66) Exception(6) tid(2008) 80070057 The parameter is incorrect.

Exception during initialization: C:\source\ort-restore\onnxruntime\onnxruntime\core\providers\dml\DmlExecutionProvider\src\MLOperatorAuthorImpl.cpp(1487)\onnxruntime.dll!00007FFB47C3F2AC: (caller: 00007FFB47C8EE66) Exception(6) tid(2008) 80070057 The parameter is incorrect.

[2021-07-15 18:29:06.074] [error] [t=8200] [K4ABT] ..\src\TrackerHost\TrackerHost.cpp (185): Create(). Exception during initialization: C:\source\ort-restore\onnxruntime\onnxruntime\core\providers\dml\DmlExecutionProvider\src\MLOperatorAuthorImpl.cpp(1487)\onnxruntime.dll!00007FFB47C3F2AC: (caller: 00007FFB47C8EE66) Exception(6) tid(2008) 80070057 The parameter is incorrect.

[2021-07-15 18:29:06.074] [error] [t=8200] [K4ABT] ..\src\sdk\k4abt.cpp (38): tracker->Create(sensor_calibration, config) returned failure in k4abt_tracker_create()
```