microsoft/Azure-Kinect-Sensor-SDK #542: Cannot create Capture in C#

rfilkov, 2019-07-25:
**Describe the bug**

I tried to create a Capture with my own color and depth data, like this:

`
                device.StartCameras(new DeviceConfiguration
                {
                    ColorFormat = ImageFormat.ColorBGRA32,
                    ColorResolution = ColorResolution.r1080p,
                    DepthMode = DepthMode.NFOV_Unbinned,
                    SynchronizedImagesOnly = true,
                });

                using (Capture capture = Sensor.Capture.Create())
                {
                    int colorWidth = 1920, colorHeight = 1080;  // ImageFormat.ColorBGRA32
                    Image colorImage = new Image(ImageFormat.ColorBGRA32, colorWidth, colorHeight, colorWidth * 4);

                    byte[] btColorImage = new byte[colorWidth * colorHeight * 4];
                    colorImage.CopyBytesFrom(btColorImage, 0, 0, btColorImage.Length);
                    colorImage.Timestamp = new TimeSpan(DateTime.Now.Ticks);

                    int depthWidth = 640, depthHeight = 576;  // DepthMode.NFOV_Unbinned
                    Image depthImage = new Image(ImageFormat.Depth16, depthWidth, depthHeight, depthWidth * sizeof(ushort));

                    byte[] btDepthImage = new byte[depthWidth * depthHeight * sizeof(ushort)];
                    depthImage.CopyBytesFrom(btDepthImage, 0, 0, btDepthImage.Length);
                    depthImage.Timestamp = new TimeSpan(DateTime.Now.Ticks);

                    capture.Color = colorImage;
                    capture.Depth = depthImage;

                    Debug.Print("Capture.Color: " + capture.Color);  // prints null
                    Debug.Print("Capture.Depth: " + capture.Depth);  // prints null
                }
`

Although colorImage and depthImage are valid, after the assignment capture.Color and capture.Depth are both null.

**To Reproduce**

Execute the code snipped above.

**Expected behavior**

I expected to get a valid Capture with valid images inside.

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  

**Additional context**

Brent-A, 2019-07-30:
Thanks @rfilkov, I see the problem. It looks like we aren't calling the k4a_capture_set_color API so the managed cache is getting overwritten the next time we read the image. I'll work on getting a fix in.

closed 2019-08-02 by Brent-A

rfilkov, 2019-08-05:
Thank you, @Brent-A

t-kurtke, 2023-06-09:
Hey! How can you do              
` depthImage.CopyBytesFrom(btDepthImage, 0, 0, btDepthImage.Length);`
?? I can't find this function anywhere

PR #570 (2019-08-02): Fix assignment of Images to Capture objects (C#)

## Fixes #542 

### Description of the changes:
- Fixed issue with assignment of Image objects to Capture in C#
- Added unit tests to cover Capture class

### Before submitting a Pull Request:
- [X] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [X] I [built my changes](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) locally
- [X] I ran the [unit tests](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/testing.md)
- [X] I  [... cut]