microsoft/Azure-Kinect-Sensor-SDK #543: Color camera image to point cloud

rfilkov, 2019-07-25:
**Is your feature request related to a problem? Please describe.**

Currently, I can't find a way to convert the color camera image to point cloud. k4a_transformation_depth_image_to_color_camera() returns an image of distances, but the XY positions in the depth image are missing. Please suggest a solution.

**Describe the solution you'd like**

A function similar to MapColorFrameToDepthSpace() in Kinect SDK 2.0 would solve this and similar issues quite elegantly. See again #502 

**Describe alternatives you've considered**

**Additional context**

cdedmonds, 2019-07-25:
@rflikov please look at k4a_transformation_depth_image_to_point_cloud. You can pass it the result of k4a_transformation_depth_image_to_color_camera to get an xyz image out.

mbleyer, 2019-07-25:
The transformation example (https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/transformation) is doing exactly what you want.

rfilkov, 2019-07-25:
@cdedmonds thank you! @mbleyer thank you, too!
I must have been blind.

closed 2019-07-25 by rfilkov