microsoft/Azure-Kinect-Sensor-SDK #547: Unable to turn off logging to stdout for body tracking runtime

bibigone, 2019-07-26:
By analogy with Sensor Tracking SDK, I've tried to use the following environment variables:
1. `K4ABT_LOG_LEVEL` 
2. `K4ABT_ENABLE_LOG_TO_STDOUT`
3. `K4ABT_ENABLE_LOG_TO_A_FILE`

The first variable works as expected.

But the second one doesn't. Setting `K4ABT_ENABLE_LOG_TO_STDOUT` to 0 (to turn off logging to stdout) results in Memory Access Violation exception in internals of Body Tracking runtime. Looks as an zero-pointer dereference when STDOUT logging is turned off.

As to the last setting, it is also not working as expected: log file is created, but there is no log messages in it. Most likely, it is connected to the impossibility to disable stdout log.

cdedmonds, 2019-07-26:
Thanks @bibigone we will look to see if we can reproduce the issue.

cdedmonds, 2019-07-31:
@bibigone I was able to reproduce the issue with the crash when disabling logging to stdout and have a fix ready for the next release. On the second issue I am able to see a log file with contents if I set the log level to trace and wait a few seconds before closing the viewer. The logger flushes irregularly unless there are errors or warnings that are logged (then it flushes immediately).

bibigone, 2019-08-01:
@cdedmonds Thank you for attention to this issue. As to:

> On the second issue I am able to see a log file with contents if I set the log level to trace and wait a few seconds before closing the viewer.

Did you try it after fixing the first issue with disabling logging to stdout? If yes, please try it before fix (on 0.9.0 version). I suppose, that these issues are connected: log file is empty in 0.9.0 version because it is impossible to turn off logging to stdout. But as soon as the first issue is fixed, logging to file starts working *if logging to stdout is disabled*.

In brief: I believe that fix you've made resolves both issues.

closed 2019-08-06 by cdedmonds