microsoft/Azure-Kinect-Sensor-SDK #550: capture error when multi-device using multi-thread to capture

YuXinFan, 2019-07-27:
**Describe the bug**
When installing multi-device (I install 4 devices), create a thread for each device to do capturing. Store captures in a queue. And the main thread gets the capture result from the queue. After running some time, the function k4a_device_get_capture() error due to capturesync_get_capture() error

**To Reproduce**

the program can get synchronized capture and runs without error.
**Logs**

 [error] [t=7504] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (271): capturesync_get_capture(device->capturesync, capture_handle, timeout_in_ms) returned failure in k4a_device_get_capture()

**Desktop (please complete the following information):**
 - OS with Version: Windows 1803
 - SDK Version:  1.1.1 

**Additional context**
This error does not occur all the time. When I run the program with gtest, it always occurs. When I run the program in PowerShell, it may occur.

YuXinFan, 2019-07-27:
**Additional information**
When I increase the number of kinect device to 6, and increase the image resolution from 720p to 2160p. An error always occurs when capturing

**Logs**
[error] [t=9128] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (143): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   2684400 type:Depth

**Screen**
[image]

wes-b, 2019-07-27:
You have too many devices on a single PC. USB can not keep up.

[image]<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F22362306%2F61991775-c1341400-b087-11e9-9e8c-eebafa108f4c.png&data=02%7C01%7CWes.Barcalow%40microsoft.com%7C2ea916a00d8d4568abae08d71268bc6a%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C636998113353824631&sdata=6F67WKe4d4KWvaEx9tNLs83gijzoRzf7%2F1vlZU1UdEE%3D&reserved=0>

YuXinFan, 2019-07-28:
For your information, I connect 4 PCIe  Data acquisitions to the PC. USBs are connected to data acquisitions. I think it may be ok.

cdedmonds, 2019-07-30:
@YuXinFan it is possible you are exceeding the available PCIe bandwidth on the machine.

wes-b, 2019-07-30:
We are happy that you had 4 devices running on a single PC.

closed 2019-07-30 by wes-b

arkader, 2021-11-02:
Hey @YuXinFan 
Did you manage to fix the issue? I've got the same when running 6 sensors at 1080p

YuXinFan, 2021-11-03:
> Hey @YuXinFan Did you manage to fix the issue? I've got the same when running 6 sensors at 1080p

@arkader I forgot the detail and I have left the project, but it has been fixed. 

I remember that this error only occurs at the starting process. It has some probability of error or normally starts in my PC.  if the device runs normally, it won't cause this error in the following running. 

Maybe you can add a mechanism in your code to catch the error and mechanism restart it until normally start.