microsoft/Azure-Kinect-Sensor-SDK #551: k4a_image_t to cv::Mat transform

nonlinear1, 2019-07-28:
I'm a newbie for Kinect for Azure. I think Kinect for Azure is great product. I think the class of  k4a_image_t is similar to cv::Mat in opencv. For many application for computer vision is coded with cv::mat of opencv. Could you add a transform function of change k4a_image_t to cv::Mat of opencv? I think this will give helps to persons who are familiar with opencv. And Let us easily change many application which are coded with cv::Mat of opencv  with Kinect for Azure sensor.

YuXinFan, 2019-07-29:
First, get the image buffer from the capture. Then,
For depth, cv::Mat mat = cv::Mat(height, width, CV_16UC1, depth_buffer, AUTO_SETP).clone();
For color, cv::Mat mat = cv::Mat(height, width, CV_8UC4, color_buffer, AUTO_SETP).clone();
Remeber to use clone() or copyTo() to get the copy of the buffer data.

nonlinear1, 2019-07-29:
Thank you very much

rabbitdaxi, 2019-07-29:
@YuXinFan Thank you for the comment. 

@nonlinear1 Thanks for the feedback, yes, I also uses opencv a lot and indeed such helper function can be really helpful. However, to keep the SDK interface clean and not introduce extra dependencies, we provide the raw buffer of the images. User has the freedom to convert the buffer to any format they like. Other than the helper that @YuXinFan kindly provided, you can also take a look this sample:
https://github.com/microsoft/Azure-Kinect-Samples/blob/master/opencv-kinfu-samples/main.cpp, and this helper also can make your ask work
'''
template<typename T> Mat create_mat_from_buffer(T *data, int width, int height, int channels = 1)
{
    Mat mat(height, width, CV_MAKETYPE(DataType<T>::type, channels));
    memcpy(mat.data, data, width * height * channels * sizeof(T));
    return mat;
}
'''

rabbitdaxi, 2019-07-29:
@nonlinear1 Let us know if you have any more questions, or we can close this issue :)

nonlinear1, 2019-07-29:
Thank you very much. I think I totally understand this issue and you can close this issue

closed 2019-07-30 by rabbitdaxi