microsoft/Azure-Kinect-Sensor-SDK #552: How to make a color image the same resolution as depth image and both are registrated?

nonlinear1, 2019-07-29:
In RGBD slams, they need that color image and depth image are both same resolution and registrated. That is to say: a pix in color  image has a counterpart pix in depth image. And at the same time, they are registrated. that is to say they are at the same angle of view. I found that in SDK no this kinds of configuration can make color image and depth image the same resolution. How can I solve this problem? For example, the resolution of depth is 640x576. How can I make configuration that can produce the color image is also 640*576?
Thanks a lot!

rabbitdaxi, 2019-07-29:
@nonlinear1 , thank you for reaching out. Yes, we already have APIs in place that can fulfill your ask. As a matter of fact, we have some examples and documentation as well. Please read the doc here: https://docs.microsoft.com/en-us/azure/Kinect-dk/use-image-transformation, also please take a look the example https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/transformation. All these APIs are GPU accelerated that transform depth and color image to each other's camera space in real time. 

Another example is the k4aviewer.exe source code, which has a mode that show color point cloud, and internally it already uses the transformation API to make this happen. 

Also, since you are interested in RGBD SLAM, our group also has a RGBD SLAM algorithm recently published, and this specific file also used our API to register color and depth images, you can search k4a_transformation_depth_image_to_color_camera, in this file: https://github.com/ETH3D/badslam/blob/17e44ed673ef8cdb6b54b953955e995f043e8a86/applications/badslam/src/badslam/input_azurekinect.cc

Please let us know whether above information addressed your issue

nonlinear1, 2019-07-29:
Thank you very much for reply and careful explanation.

closed 2019-07-30 by rabbitdaxi