microsoft/Azure-Kinect-Sensor-SDK #553: Retrieve image raw data

nicolascarre, 2019-07-29:
I'm currently using the MJPG format, but I also need the raw data in order to compute the luminance, is exists a method or something to get the raw data from the image ?

rabbitdaxi, 2019-07-29:
@nicolascarre thanks for your feedback, please take a look [k4a_image_format_t](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___enumerations_gabd9688eb20d5cb878fd22d36de882ddb.html#gabd9688eb20d5cb878fd22d36de882ddb), and specifically K4A_IMAGE_FORMAT_COLOR_NV12 and K4A_IMAGE_FORMAT_COLOR_YUY2, where you can find the luminance and chroma data. Please let us know whether this addressed your issue

nicolascarre, 2019-07-29:
Yes, but I'm using the K4A_IMAGE_FORMAT_COLOR_MJPG format, so your way can't works for me, or I need twice...

rabbitdaxi, 2019-07-29:
@nicolascarre , what about you use the NV12 or YUY2 format to get the image (then you have the luminance info), and you can convert to bgra image? k4aviewer does such conversion as well: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/tools/k4aviewer/k4acolorimageconverter.cpp, see K4AYUY2ImageConverter or K4ANV12ImageConverter

nicolascarre, 2019-07-29:
Yes but I need to save the image in a file at the .jpeg format, I have the best performance with MJPG, because this imediatly in jpeg format, no need any transformation.

rabbitdaxi, 2019-07-29:
@nicolascarre , unfortunately, the data we get form the color camera firmware either MJPG, or NV12 and YUY2. If you are consuming MJPG and you want to get luminance, you can decode the MJPG to YUV, e.g. libjpeg-turbo?

cdedmonds, 2019-08-06:
It seems like this issue is resolved. Closing.

closed 2019-08-06 by cdedmonds