microsoft/Azure-Kinect-Sensor-SDK #555: Error while trying to capture depth imagery from multiple synced devices

darahan, 2019-07-29:
Hi,

I try to record a point cloud with multiple images in sync. However I get the following set of errors:
`[2019-07-29 15:43:52.561] [error] [t=13600] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (46): k4a_image_t_get_context(). Invalid k4a_image_t 0000000000000000
[2019-07-29 15:43:52.569] [error] [t=13600] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (249): Invalid argument to image_get_width_pixels(). image_handle (0000000000000000) is not a valid handle of type k4a_image_t
[2019-07-29 15:43:52.569] [error] [t=13600] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (46): k4a_image_t_get_context(). Invalid k4a_image_t 0000000000000000
[2019-07-29 15:43:52.569] [error] [t=13600] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (256): Invalid argument to image_get_height_pixels(). image_handle (0000000000000000) is not a valid handle of type k4a_image_t
[2019-07-29 15:43:52.569] [error] [t=13600] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (46): k4a_image_t_get_context(). Invalid k4a_image_t 0000000000000000
[2019-07-29 15:43:52.569] [error] [t=13600] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (218): Invalid argument to image_get_buffer(). image_handle (0000000000000000) is not a valid handle of type k4a_image_t

[2019-07-29 15:43:53.332] [error] [t=13600] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (46): k4a_image_t_get_context(). Invalid k4a_image_t 0000000000000000
[2019-07-29 15:43:53.332] [error] [t=13600] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (192): Invalid argument to image_dec_ref(). image_handle (0000000000000000) is not a valid handle of type k4a_image_t

[2019-07-29 15:43:53.332] [error] [t=13600] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (46): k4a_image_t_get_context(). Invalid k4a_image_t 0000000000000000
[2019-07-29 15:43:53.332] [error] [t=13600] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (249): Invalid argument to image_get_width_pixels(). image_handle (0000000000000000) is not a valid handle of type k4a_image_t
[2019-07-29 15:43:53.332] [error] [t=13600] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (46): k4a_image_t_get_context(). Invalid k4a_image_t 0000000000000000
[2019-07-29 15:43:53.332] [error] [t=13600] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (256): Invalid argument to image_get_height_pixels(). image_handle (0000000000000000) is not a valid handle of type k4a_image_t
[2019-07-29 15:43:53.332] [error] [t=13600] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (46): k4a_image_t_get_context(). Invalid k4a_image_t 0000000000000000
[2019-07-29 15:43:53.332] [error] [t=13600] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (218): Invalid argument to image_get_buffer(). image_handle (0000000000000000) is not a valid handle of type k4a_image_t

[2019-07-29 15:43:53.911] [error] [t=13600] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (46): k4a_image_t_get_context(). Invalid k4a_image_t 0000000000000000
[2019-07-29 15:43:53.911] [error] [t=13600] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\image\image.c (192): Invalid argument to image_dec_ref(). image_handle (0000000000000000) is not a valid handle of type k4a_image_t`
And here is my code:
```C++
#include <array>
#include <iostream>
#include <fstream>
#include <sstream>
#include <k4a/k4a.hpp>
#include <math.h>

// Licensed under the MIT License.

static void create_xy_table(const k4a_calibration_t *calibration, k4a_image_t xy_table)
{
    k4a_float2_t *table_data = (k4a_float2_t *)(void *)k4a_image_get_buffer(xy_table);

    int width = calibration->depth_camera_calibration.resolution_width;
    int height = calibration->depth_camera_calibration.resolution_height;

    k4a_float2_t p;
    k4a_float3_t ray;
    int valid;

    for (int y = 0, idx = 0; y < height; y++)
    {
        p.xy.y = (float)y;
        for (int x = 0; x < width; x++, idx++)
        {
            p.xy.x = (float)x;

            k4a_calibration_2d_to_3d(
                calibration, &p, 1.f, K4A_CALIBRATION_TYPE_DEPTH, K4A_CALIBRATION_TYPE_DEPTH, &ray, &valid);

            if (valid)
            {
                table_data[idx].xy.x = ray.xyz.x;
                table_data[idx].xy.y = ray.xyz.y;
            }
            else
            {
                table_data[idx].xy.x = nanf("");
                table_data[idx].xy.y = nanf("");
            }
        }
    }
}

static void generate_point_cloud(const k4a_image_t depth_image,
                                 const k4a_image_t xy_table,
[... 180 lines cut ...]
        depth_images[i]= k4a_capture_get_depth_image(capture.get_handle());
        capture.reset();

    }

    for(size_t i = 0; i < devices.size(); i++){
        size_t deviceNo = i + 1;
        int pointCount = 0;
        file_names[i] += argv[1];
        file_names[i] += "Camera"+std::to_string(deviceNo)+".ply";

        generate_point_cloud(depth_images[i], xy_tables[i], point_clouds[i], &pointCount);
        write_point_cloud(file_names[i].c_str(), point_clouds[i], pointCount);
        k4a_image_release(depth_images[i]);
        k4a_image_release(point_clouds[i]);
        k4a_image_release(xy_tables[i]);
    }

    close_devices(devices);

    return 0;
}

```
I added a get_handle to the calibration class in the C++ wrapper in order to obtain the corresponding handle, Why do I get this error?

spschul, 2019-07-29:
In order to guarantee that a capture from a camera arrives with both color and depth, you must set the `synchronized_images_only` flag on `k4a_device_configuration_t`. Without that flag, you are likely receiving valid captures that only have a color image stored within them, and so `k4a_capture_get_depth_image` will not be able to get a depth image.

I'm closing this issue, because this is very likely the cause, but if the problem persists please comment here to let us know!

closed 2019-07-29 by spschul

opalmu, 2019-08-20:
@spschul Thanks for your information :-). I received this message as well when I was reading images from recorded data in ```.mkv``` file.  In my code I only received depth images properly using ```k4a_capture_get_depth_image```. When I tried to read colored images with ``` k4a_capture_get_color_image``` it throws me the error:
``` [2019-08-19 17:24:49.017] [error] [t=1735] /home/vsts/work/1/s/extern/Azure-Kinect-Sensor-SDK/src/image/image.c (192): Invalid argument to image_dec_ref(). image_handle ((nil)) is not a valid handle of type k4a_image_t```

I checked that there is no input variables to be fed into ```k4a_playback_open```, ```k4a_playback_get_next_capture``` and ```k4a_capture_get_depth_image```. Do you happen to know how could I solve this issue?  Here I also attached my code. Many thanks!
```
#include <stdio.h>
#include <stdlib.h>
#include <k4a/k4a.h>
#include <k4a/k4atypes.h>
#include <k4arecord/playback.h>

int main()
{
	// Open the .mkv file
	k4a_playback_t playback_handle = NULL;
	const char* filePath = "/home/ryan/C_program/output.mkv";
	if (k4a_playback_open(filePath, &playback_handle) != K4A_RESULT_SUCCEEDED)
	{
    	    printf("Failed to open recording\n");
    	    return 1;
	}
	// Start to read the captures
	k4a_capture_t capture = NULL;
	k4a_stream_result_t result = K4A_STREAM_RESULT_SUCCEEDED;

	// Set the device configuration
	k4a_device_configuration_t deviceConfig = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
	deviceConfig.synchronized_images_only = true;

	while (result == K4A_STREAM_RESULT_SUCCEEDED)
	{
	    result = k4a_playback_get_next_capture(playback_handle, &capture);
	    if (result == K4A_STREAM_RESULT_SUCCEEDED)
	    {
	        // PROCESS THE CAPTURES HERE
		// Read images from capture
	        k4a_image_t image_color = k4a_capture_get_color_image(capture);
		k4a_image_t image_depth = k4a_capture_get_depth_image(capture);

		// Get width and height
		int width = k4a_image_get_width_pixels(image_color);
		int height = k4a_image_get_height_pixels(image_color);
		int strides = k4a_image_get_stride_bytes(image_color);
		printf("Image width is %d\n", width);
		printf("Image height is %d\n", height);
		printf("Image strides is %d\n", strides);
		
		// Create the buffer to read images
		uint8_t * imgData_read = k4a_image_get_buffer(image_color);
		uint8_t * imgData_write = (uint8_t*)malloc(strides*height*sizeof(uint8_t));
		for (int i = 0; i < strides * height; i++)
		{
		    uint16_t pixelValue = *((uint16_t*)(imgData_read + i));
		    uint8_t value = static_cast<uint8_t>((pixelValue - 0) * (double(65535) / (65535- 0)));
		    printf("Value is %d\n", value);
		    imgData_write[i] = value;
		}
		// Call k4a_image_release when finished with this image.
	        k4a_image_release(image_color);
		k4a_image_release(image_depth);

	        // Call k4a_capture_release when finished using the capture.
	        k4a_capture_release(capture);
	    }
	    else if (result == K4A_STREAM_RESULT_EOF)
	    {
	        printf("The result is K4A_STREAM_RESULT_EOF...");
	        break;
	    }
	    else if (result == K4A_STREAM_RESULT_FAILED)
	    {
	        printf("The result is K4A_STREAM_RESULT_FAILED...");
	    }

	}
	// Failed to read captures
	if (result == K4A_STREAM_RESULT_FAILED)
	{
	    printf("Failed to read entire recording\n");
       	    return 1;
	}

	// Close the reading of .mkv file
	k4a_playback_close(playback_handle);
	return 0;
}
```

RoseFlunder, 2019-08-20:
The device config is meant to be passed into the `k4a_device_start_cameras` method.
So when you are using a real device and not a recording.

The recording tool delivered with the SDK has no option to capture only synchronized images as far as I can see.
So you have to check which images the current capture from the recording contains.
For example if you only want to process captures which got at color image you have to check if the color image in the capture is not a nullptr:
```
// Skip captures without a color image inside the while loop
k4a_image_t image_color = k4a_capture_get_color_image(capture);
if (image_color == nullptr)
{
  // continue with next capture
  continue;
}
```

opalmu, 2019-08-20:
@ RoseFlunder Thanks for your clarification! I am switching to process with the data received from the camera in the real time since my project needs to work with both the depth images and RGB images simultaneously. But working with both of these two kinds of images with the recorded file should be fine as well. This is a very straightforward functionality in realsense D400 series. 
Anyway thanks again for your help.