microsoft/Azure-Kinect-Sensor-SDK #557: k4arecorder capability to record only frames with synchronized depth/color frames

mffigueroa, 2019-07-29:
**Is your feature request related to a problem? Please describe.**
In k4arecorder, there is currently no command-line option to record only frames with both depth and color frames received. I.e. to set the synchronized_images_only option in k4a_device_configuration_t.

**Describe the solution you'd like**
Add a --synchronized_images_only command-line option to set this option when recording a clip.

wes-b, 2019-07-30:
Why would the recorder need this? My thinking is that the recorder should record what it gets. Playback APIs should have the functionality to only deliver synchronized frames.

cdedmonds, 2019-07-30:
I think the problem is that the playback API does not have this functionality currently.

wes-b, 2019-07-30:
Gotcha, @xthexder how about we add k4a_playback_set_sychronized_images_only()?

xthexder, 2019-07-30:
I think this was originally a feature, but was cut due to confusion over what the behavior should be.
If a recording only has color enabled, or only depth/ir, would the playback API drop every capture? Or should we still return the single track?
I'm not sure what the current behavior of `synchronized_images_only` is when the camera is only running a single stream.