microsoft/Azure-Kinect-Sensor-SDK #561: Add C++ wrapper for Azure Kinect Body Tracking SDK

UnaNancyOwen, 2019-07-30:
I think we need to add C++ wrapper for Azure Kinect Body Tracking SDK.
Azure Kinect Sensor SDK already has C++ wrappers (k4a/k4a.hpp, k4arecord/playback.hpp).
However, It doesn't exist in Azure Kinect Body Tracking SDK yet.
Do you have any plan to add k4abt.hpp? Thanks,

RoseFlunder, 2019-08-05:
Is there an estimated time of arrival for the c++ wrapper?
I want to extend the azure_kinect_ros_driver to publish the Joint positions as a ROS Marker Array.
The ros package uses the c++ sensor sdk api and I don't see a way to get the C-structs from it to use it with the C-API from the body tracking SDK.

UnaNancyOwen, 2019-08-05:
@RoseFlunder I think your problem is rerated this issue #416.

RoseFlunder, 2019-08-05:
@UnaNancyOwen
Yes you are right. Thats exactly what I would need to use body tracking sdk.
Either acceess to the private struct inside of the capture class or a c++ wrapper from the Body Tracking SDK that accepts a capture reference from c++ Sensor sdk api

UnaNancyOwen, 2019-08-12:
I was pre-implement this wrapper. I want a wrapper like this. Thanks,
https://gist.github.com/UnaNancyOwen/41b578f5d272aa6f22cf8ff6565fb71b#file-k4abt-hpp

yijiew, 2019-08-12:
Thanks @UnaNancyOwen for your sample implementation. We will provide this C++ wrapper once the #626 change in the Sensor SDK is published.

UnaNancyOwen, 2019-08-23:
@yijiew Azure Kinect Sensor SDK v1.2.0 that #626 merged has been released. I look forward to official C++ wrapper being published! Thanks, 👍

yijiew, 2019-08-23:
Thanks for informing us! We will publish an official C++ wrapper in our next release. Please stay tuned :)

yijiew, 2019-08-30:
The new release 0.9.2 is published! The official C++ wrapper is published. You can download this now at: [msi](https://www.microsoft.com/en-us/download/details.aspx?id=100128),  [nuget](https://www.nuget.org/packages/Microsoft.Azure.Kinect.BodyTracking/0.9.2) or apt-get on Linux.

closed 2019-08-30 by yijiew

scimad, 2020-11-02:
> The new release 0.9.2 is published! The official C++ wrapper is published. You can download this now at: [msi](https://www.microsoft.com/en-us/download/details.aspx?id=100128), [nuget](https://www.nuget.org/packages/Microsoft.Azure.Kinect.BodyTracking/0.9.2) or apt-get on Linux.

Is it available on Ubuntu 20.04?