microsoft/Azure-Kinect-Sensor-SDK #563: Unable to get color images

wuneng, 2019-07-30:
**Describe the bug**

Unable to get color images from 21th frame
**To Reproduce**

    k4a_device_configuration_t device_config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
    device_config.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;
	device_config.color_resolution = K4A_COLOR_RESOLUTION_720P;
	device_config.synchronized_images_only = false;
    k4a_device_t device;
    VERIFY(k4a_device_open(0, &device), "Open K4A Device failed!");
    VERIFY(k4a_device_start_cameras(device, &device_config), "Start K4A cameras failed!");

    k4a_calibration_t sensor_calibration;
    VERIFY(k4a_device_get_calibration(device, device_config.depth_mode, device_config.color_resolution , &sensor_calibration),
        "Get depth camera calibration failed!");

    k4abt_tracker_t tracker = NULL;
    VERIFY(k4abt_tracker_create(&sensor_calibration, &tracker), "Body tracker initialization failed!");

    int frame_count = 0;
    do
    {
        k4a_capture_t sensor_capture;
        k4a_wait_result_t get_capture_result = k4a_device_get_capture(device, &sensor_capture, K4A_WAIT_INFINITE);
        if (get_capture_result == K4A_WAIT_RESULT_SUCCEEDED)
        {
            frame_count++;

            printf("Start processing frame %d\n", frame_count);

            k4a_wait_result_t queue_capture_result = k4abt_tracker_enqueue_capture(tracker, sensor_capture, K4A_WAIT_INFINITE);
			k4a_capture_release(sensor_capture);
            if (queue_capture_result == K4A_WAIT_RESULT_TIMEOUT)
            {
                // It should never hit timeout when K4A_WAIT_INFINITE is set.
                printf("Error! Add capture to tracker process queue timeout!\n");
                break;
            }
            else if (queue_capture_result == K4A_WAIT_RESULT_FAILED)
            {
                printf("Error! Add capture to tracker process queue failed!\n");
                break;
            }

            k4abt_frame_t body_frame = NULL;
            k4a_wait_result_t pop_frame_result = k4abt_tracker_pop_result(tracker, &body_frame, K4A_WAIT_INFINITE);
			k4a_capture_t capture_result = k4abt_frame_get_capture(body_frame);
			k4a_image_t color_image = k4a_capture_get_color_image(capture_result);

			if (pop_frame_result == K4A_WAIT_RESULT_SUCCEEDED)
            {
                size_t num_bodies = k4abt_frame_get_num_bodies(body_frame);
                printf("%zu bodies are detected!\n", num_bodies);

                for (size_t i = 0; i < num_bodies; i++)
                {
					
                    k4abt_body_t body;
                    VERIFY(k4abt_frame_get_body_skeleton(body_frame, i, &body.skeleton), "Get body from body frame failed!");
                    body.id = k4abt_frame_get_body_id(body_frame, i);
					
                    print_body_information(body);
                }

                k4abt_frame_release(body_frame);
            }
            else if (pop_frame_result == K4A_WAIT_RESULT_TIMEOUT)
            {
                //  It should never hit timeout when K4A_WAIT_INFINITE is set.
[... 155 lines cut ...]
[2019-07-30 18:19:07.068] [error] [t=3464] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (143): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   2949422 type:Depth
2 bodies are detected!
Body ID: 2
Body ID: 1
[2019-07-30 18:19:07.098] [error] [t=3464] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (143): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   2982755 type:Depth
Start processing frame 29
[2019-07-30 18:19:07.132] [error] [t=3464] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (143): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   3016088 type:Depth
2 bodies are detected!
Body ID: 1
Body ID: 2
Start processing frame 30
[2019-07-30 18:19:07.165] [error] [t=3464] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (143): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   3049422 type:Depth
2 bodies are detected!
Body ID: 2
Body ID: 1
Finished body tracking processing!

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version:  WIN10
 - SDK Version:   1.1.0
 - Firmware version:  

**Additional context**

rabbitdaxi, 2019-07-30:
@wuneng , this is by design I think. Because you set device_config.synchronized_images_only = false; please read the document about [synchronized_images_only](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/structk4a__device__configuration__t_a8208974f05d89fc1362c6a0900bdef4d.html#a8208974f05d89fc1362c6a0900bdef4d) option. If you want always have both color and depth available in the capture, you need to set this flag to true. If not, you may get a capture contains only depth or color.

wes-b, 2019-07-30:
device_config.color_format is not set and still initialized too off. So the sensor is not delivering color frames

closed 2019-07-30 by wes-b

wes-b, 2019-07-30:
``` Unable to get color images from 21th frame```

Looks like captures and images are not being released properly. You are hitting a Media Foundation implementation limit on the number of color captures that can be outstanding at one time. Make sure you call k4a_capture_release and k4a_image_release when you are done working with those objects.