microsoft/Azure-Kinect-Sensor-SDK #565: Get depth value at a pixel

snair61, 2019-07-30:
I am tying to get depth value at a pixel of the color image. Since the depth and color image have different resolutions, I am trying to use function `k4a_transformation_color_image_to_depth_camera` to get the depth pixel. Is this the right approach?

`k4a_image_create( K4A_IMAGE_FORMAT_COLOR_BGRA32, 
				depth_image_width_pixels,depth_image_height_pixels,
					k4a_image_get_stride_bytes(depth_image),
					 &transformed_color_image)

k4a_transformation_color_image_to_depth_camera(transform, depth_image, color_image, transformed_color_image`

I get the error:

[error] [t=4167] /home/vsts/work/1/s/extern/Azure-Kinect-Sensor-SDK/src/transformation/transformation.c (534): transformation_color_image_to_depth_camera(). Expect both depth camera and color camera are running to transform color image to depth camera.
Currently, I m using one kinect device and the configuration for my camera is mentioned below from which I capture both the color and depth images. How can I get the depth value at a pixel of a color image?

My camera configuration is given below:

`k4a_device_configuration_t config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
	config.camera_fps = K4A_FRAMES_PER_SECOND_30;
	config.color_format = K4A_IMAGE_FORMAT_COLOR_BGRA32;
	config.color_resolution = K4A_COLOR_RESOLUTION_2160P;
	config.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;
	config.synchronized_images_only=true;

// Start the camera with the given configuration

if (K4A_RESULT_SUCCEEDED != k4a_device_start_cameras(device, &config))`

rabbitdaxi, 2019-07-30:
@snair61 , please take a look the example https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/transformation, also please take a look the documentation: https://docs.microsoft.com/en-us/azure/Kinect-dk/use-image-transformation. Let us know whether the example can help you solve your problem.

snair61, 2019-07-30:
Thank you. That resolved it. My sensor calibration was not right.

closed 2019-07-30 by snair61

Hujustin, 2019-11-26:
@rabbitdaxi 
Hi, if I want to show the depth camera with gray mode.
such as kinect v2 we can see the depth but only in gray scale.
I just bought it . And I used to use kincet v2 in unity only.

fatihoezdemir, 2020-09-14:
Hi @snair61, how did you solve the sensor calibration problem ?