microsoft/Azure-Kinect-Sensor-SDK #568: Getting Depth engine for arm64

lullabee, 2019-07-31:
We are looking into using the kinect on robots using TX2, Jetpack 4.2.1
How can we request a compiled version of  the Depth Engine for this version? The rest of the engine seems to cross compile fine.

nonlinear1, 2019-07-31:
I am also eager to get  the Detph Engine for ARM 64 for Jetson Xavier. It seems that they have not decided to compile the depth engine for arm64.

cdedmonds, 2019-07-31:
This is one of our top user voice feedback items. Please go upvote there. https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/37615903-arm-support-for-linux

closed 2019-07-31 by cdedmonds

eric-schleicher, 2019-08-03:
Cool!  so now that it's established that this is the #1 feedback item/request...   any forecast on how long until we might see this?