microsoft/Azure-Kinect-Sensor-SDK #575: Body orientation flip when wearing glasses

PierrePlantard, 2019-08-02:
Hello, we have an issue with the body tracker when we wearing an hood and protection glasses (see image below). In these case the body constantly flip. 
I thing this come from the difficulty of the body tracker to well measure the eyes and/or the ears.
As you can see on the enclosed video, the skeleton id change often and the skeleton is flipped to 180°. This phenomenon was only found for some people of my team, not for all people.
This is a huge issue in industrial or health applications because most people have such protection equipment or hygienic cloths. 

**To Reproduce**
1. Tested in Simple_3D_Viewer sample
2. Wear protection glasses or other 

**Expected behavior**
The global orientation stability need to be improved. 
Some ideas : 
 - Detecting flip and correct it.
- Take another body part to compute the global orientation.
- Improve the eyes and ears tracking.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS with Version: OS with Version: Windows 10
 - SDK Version:  1.0.1

Brekel, 2019-08-02:
I've seen these kind of flips (as well as the tracking being lost and being picked up as another body ID as can bee seen in your clip) on occasions when not wearing glasses too.

cdedmonds, 2019-08-06:
Thank you for reporting this issue. We are investigating fixes to make flipping less common.

PierrePlantard, 2019-08-06:
Thank you for your response.
Keep us in touch.

Best regards,

Pierre

RenderHeadsMrT, 2019-09-16:
Hi, we've been testing the Azure Kinect for a while now and it seems the biggest issue we're facing is this random flipping (SDK 1.2.0, tracker 0.9.2, 'K4A_DEPTH_MODE_NFOV_UNBINNED', 'K4A_IMAGE_FORMAT_COLOR_BGRA32', 'K4A_COLOR_RESOLUTION_720P').

Due to our use case we either have to mount low (~floor height) or high (~2.3m), tilted up/down to track a person standing ~2.5m away. Our preference is low as it seems to provide better tracking of smaller people/children and fewer issues with perspective/lean.

The problem we're seeing is that in the low position this flip issue is occurring quite frequently, often when just standing still. It seems leaning slightly towards the camera (showing it more face) helps but is not a practical fix. In the high position this flipping is far less prevalent, I'd presume down to more face being visible, but instead opens up other issues such as not tracking shorter people and more acute angles to compensate for.

I'm currently looking into trying to catch the flip somehow and filter it out but wondered if you could provide any updates on this issue or any further guidance on how to minimise it?

Cheers,
Richard

PierrePlantard, 2019-11-21:
Hi,
Any update relative to this bug ?
It is also the biggest issue we are facing. 
I can send you some example in video to illustrate the problem (in private).
Keep us in touch,

Cheers, 
Pierre

jesd23, 2019-11-21:
N^ Hot. Pick wisely please.

jesd23, 2019-11-21:
Blu easy to near grn two not one malt .

qm13, 2019-11-22:
Could you please send us a recording illustrating the issue.

PierrePlantard, 2019-11-22:
Thank you,
do you have an email to send you the video ?
Pierre

PierrePlantard, 2019-11-25:
Dear @qm13, please give me an email to send the video.

Regards,
Pierre

qm13, 2019-12-03:
@PierrePlantard  we would like to get a recording that demonstrates the problems you are reporting. We require a signed data release agreement along with the recording. Could you provide me with contact details so I can provide you both the agreement and upload instructions?

PierrePlantard, 2019-12-04:
@qm13 yes, you can email me at pierre.plantard@moovency.com
Thanks

PierrePlantard, 2019-12-11:
Dear @qm13, I still haven't received your email, is it normal ?

Chrisbright10, 2020-03-08:
This issue is a big one for me as well, I am not able to adopt this camera at this time because of the body flipping. It doesn't seem to be just isolated to the glasses, (please note issue #1074 ). I have had this issue often while trying to implement this camera in my current project. Could we please prioritize this issue? It is the main reason I can't adopt the Azure Kinect DK camera in my current product (and I would so love to switch over to this).

For the record I am using SDK 1.3.0, tracker 1.0.0, 'K4A_DEPTH_MODE_NFOV_UNBINNED', 'K4A_IMAGE_FORMAT_COLOR_BGRA32', 'K4A_COLOR_RESOLUTION_720P'

Might I ask (without diving too far into the codebase)...  how are we currently determining the orientation of the body in body-tracking mode?

Chrisbright10, 2020-06-04:
Is there any update on this issue? If not, who can I pay to help fix this issue faster? This is the only issue preventing me from adopting this camera at this time, and I would really love to.

brontesprocessing, 2020-12-12:
I have exactly the same problem - when I wear glasses the problem occurs very often (every several seconds).