microsoft/Azure-Kinect-Sensor-SDK #576: Runtime out of memory error

Shri-Humrudha, 2019-08-02:
Bug description
Cloud points are not viewed in Body tracking viewer

Steps to reproduce the behavior:
1. Go to body tracking viewer
2. Infra image appears
3. Run time error is printed in the command window
C:\repos\onnxruntime\onnxruntime\core\providers\cuda\cuda_call.cc:97 onnxruntime:cudacall c:\repose\onnxruntime\onnxruntime\core\providers\cuda\cuda_call.cc:91 onnxruntime::cudacall CUDS failure 2: out of memory ; GPU=9; hostname=<machinename> ; expr=cudamalloc((void**)&p, size);
4. No cloud points of the skeleton

Screenshot:
[image]

Desktop 
 - OS with Version: Windows 10 Enterprise 1903 18362.239
 - SDK Version:  0.9.0
 - Firmware version:  
 == Azure Kinect DK Firmware Tool ==
Device Serial Number: 000212292212
Current Firmware Versions:
  RGB camera firmware:      1.6.98
  Depth camera firmware:    1.6.70
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

cdedmonds, 2019-08-06:
@Shri-Humrudha it sounds like your GPU does not have enough free memory to run body tracking. Do you have other applications open on your machine?

YZR821, 2019-08-15:
Hi, I got some problems like this too! I wanna to run my programs for 4 times to get joints information from four different .mvk files, however when processing the fourth video, the cmd line gave me an error information like yours and I have no idea how to solve it. If you found a solution and can share it with me, I'll be appreciate that!

cdedmonds, 2019-08-15:
@YZR821 and @Shri-Humrudha it sounds like your GPUs are running out of available GPU memory for body tracking. This could be due to opening multiple trackers or other programs that are running.

closed 2019-08-15 by cdedmonds

Shri-Humrudha, 2019-08-15:
I have only one tracker application opened. It was never working for me
though.

cdedmonds, 2019-08-15:
Can you please comment on what GPU you are using?

YZR821, 2019-08-16:
@cdedmonds First, let me explain my project to you: I run a piece of code for four time, and each time I create one tracker to get the joints information from a .mkv file. After finishing one .mkv file, I shutdown and destroyed the tracker, then I open another .mkv file and do the same things. However, I noticed that my  GPU memory didn't release after I shutdown and destroy the tracker, and I didn't find any other functions in the SDK to solve this problem. I'm not sure whether I explained my situation clearly, but thanks for your help!