microsoft/Azure-Kinect-Sensor-SDK #577: C# Wrapper DLL Files

Archetype90, 2019-08-02:
I am attempting to use the C# wrapper. I have downloaded the develop branch, added the solution file to my project, and added a reference to my project. This allowed me to use Microsoft.Azure.Kinect.Sensor and all methods. But, when attempting to run the program, I was getting an error:

> Unable to load DLL 'k4a'

Fairly explicit error. I had installed the MSI previously and added the folder including the k4a.dll, k4a.pdb, and depthengine_1_0.dll to my %PATH%. This led to a new error:

> An attempt was made to load a program with an incorrect format.

Less explicit error, but I assume the DLL generated using the MSI is not compatible with the solution that I imported. 

Where should I be getting the DLL to use the C# code from the develop branch? Or, if this is not the correct question - what am I doing wrong?

Archetype90, 2019-08-02:
OK, silly mistake by me. I was building against "Any CPU" vs. x64. Adding a new configuration of x64 fixed the above error of attempting to load a program with an incorrect format. But, I am onto a new issue:

> System.TypeInitializationException: 'The type initializer for 'Microsoft.Azure.Kinect.Sensor.Allocator' threw an exception.'

As above, I am not really sure where I am going wrong.

Brent-A, 2019-08-02:
Hi @Archetype90. If you'd like to get started with the C# wrapper, I'd recommend getting it from the [prerelease nuget package](https://www.nuget.org/packages/Microsoft.Azure.Kinect.Sensor/1.2.0-alpha.10). That has all the matching binaries published together and it should bring in the right set of dependencies you need.

There have been some new features added to k4a.dll and the depthengine that are needed by the C# wrapper in develop.

closed 2019-08-02 by Brent-A

Archetype90, 2019-08-02:
Thanks a ton @Brent-A. I have installed the prerelease nuget package and removed all references to the previous solution and DLL files (removed from %PATH%). Unfortunately, I am seeing the same problem as above. 

When Device.Open() is called it throws the InitializationException. I have confirmed that I am able to open and view the Kinect when using the k4aviewer.exe, so it is not some silly issue like not having it plugged in.

Brent-A, 2019-08-05:
I see. It looks like the nuget package is missing some metadata for the dependencies. If you add System.Buffers as a dependency to your project it will probably start working.