microsoft/Azure-Kinect-Sensor-SDK #582: [Proposal] Create output k4a::image in k4a::transformation

UnaNancyOwen, 2019-08-03:
I propose that create output memory in k4a::transformation.
It is no necessary to write code to create output image by developer.
Therefor, It will shortens the developer side code.

We can implement it with little changes in k4a.hpp.

```patch
     /** Transforms the depth map into the geometry of the color camera.
      * Throws error on failure
      *
      * \sa k4a_transformation_depth_image_to_color_camera
      */
-    void depth_image_to_color_camera(const image &depth_image, image *transformed_depth_image) const
+    void depth_image_to_color_camera(const image &depth_image, 
+                                     const image &color_image,
+                                     image *transformed_depth_image) const
     {
+        if (transformed_depth_image)
+        {
+            *transformed_depth_image = image::create(K4A_IMAGE_FORMAT_DEPTH16,
+                                                     color_image.get_width_pixels(),
+                                                     color_image.get_height_pixels(),
+                                                     color_image.get_width_pixels() * static_cast<int32_t>(sizeof(uint16_t)));
+        }

         k4a_result_t result = k4a_transformation_depth_image_to_color_camera(m_handle,
                                                                              depth_image.handle(),
                                                                              transformed_depth_image->handle());
         if (K4A_RESULT_SUCCEEDED != result)
         {
             throw error("Failed to convert depth map to color camera geometry!");
         }
     }

     /** Transforms the color image into the geometry of the depth camera.
      * Throws error on failure
      *
      * \sa k4a_transformation_color_image_to_depth_camera
      */
     void color_image_to_depth_camera(const image &depth_image,
                                      const image &color_image,
                                      image *transformed_color_image) const
     {
+        if (transformed_color_image)
+        {
+            *transformed_color_image = image::create(K4A_IMAGE_FORMAT_COLOR_BGRA32,
+                                                     depth_image.get_width_pixels(),
+                                                     depth_image.get_height_pixels(),
+                                                     depth_image.get_width_pixels() * 4 * static_cast<int32_t>(sizeof(uint8_t)));
+        }

         k4a_result_t result = k4a_transformation_color_image_to_depth_camera(m_handle,
                                                                              depth_image.handle(),
                                                                              color_image.handle(),
                                                                              transformed_color_image->handle());
         if (K4A_RESULT_SUCCEEDED != result)
         {
             throw error("Failed to convert color image to depth camera geometry!");
         }
     }

     /** Transforms the depth image into 3 planar images representing X, Y and Z-coordinates of corresponding 3d points.
      * Throws error on failure.
      *
      * \sa k4a_transformation_depth_image_to_point_cloud
      */
     void depth_image_to_point_cloud(const image &depth_image, k4a_calibration_type_t camera, image *xyz_image) const
     {
+        if (xyz_image)
+        {
+            *xyz_image = image::create(K4A_IMAGE_FORMAT_CUSTOM,
+                                       depth_image.get_width_pixels(),
+                                       depth_image.get_height_pixels(),
+                                       depth_image.get_width_pixels() * 3 * static_cast<int32_t>(sizeof(int16_t)));
+        }

         k4a_result_t result =
             k4a_transformation_depth_image_to_point_cloud(m_handle, depth_image.handle(), camera, xyz_image->handle());
         if (K4A_RESULT_SUCCEEDED != result)
         {
             throw error("Failed to transform depth image to point cloud!");
         }
     }

```
I think that is no problem in general use case. And, I hope that will helpful for developers.
What do you think? Thanks,

Brent-A, 2019-08-05:
Thanks @UnaNancyOwen,

We think this is a good idea. We are planning something similar for the C# wrapper (https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/src/csharp/SDK/Transformation.cs).

A couple of pieces of feedback:

1. Let's not pass in the extra color image just to get the resolution. Instead we can get the appropriate resolution during the constructor for k4a::transformation and store the resolution as member variables of the transformation class. These values are available in the calibration object that was used to initialize the k4a::transformation object.

2. Instead of making this an output parameter, we can just return the new k4a::image as the return argument in the case where the k4a::transformation creates the image. I think this is a bit cleaner since we now have two distinct overloads, one in which an existing image is modified, and another where it is allocated.

Brent-A, 2019-08-05:
I'm marking this as approved. If you would like to submit a pull request, we can go in to more detailed review in a PR.

UnaNancyOwen, 2019-08-06:
@Brent-A Thanks! I created a pull request #596.

PR #596 (2019-08-08): Add overload functions that returns k4a::image in k4a::transformation
This pull request will implements #582.
That proposal is add following two features related <code>k4a::transformation</code>.
(There are already implemented similar features in C# API.)

* [x] Creates output <code>k4a::image</code> in <code>k4a::transformation</code>.
* [x] Add overload functions that returns <code>k4a::image</code>.

I think this change will helpfull developers by simplifies user code. It make simple.
(It doesn't break existing user code by this change. It can be using the legacy user code as it is.)
```cpp [... cut]
closed 2019-08-08 by Brent-A