microsoft/Azure-Kinect-Sensor-SDK #585: Can't get color data

riverMountainjs, 2019-08-05:
`// Copyright (c) Microsoft Corporation. All rights reserved.
// Licensed under the MIT License.

#include <stdio.h>
#include <stdlib.h>
#include <k4a/k4a.h>

int main()
{
	int returnCode = 1;
	k4a_device_t device = NULL;
	const int32_t TIMEOUT_IN_MS = 1000;
	int captureFrameCount;
	k4a_capture_t capture = NULL;
	k4a_device_configuration_t config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;

	captureFrameCount = 300;
	printf("Capturing %d frames\n", captureFrameCount);

	uint32_t device_count = k4a_device_get_installed_count();

	if (device_count == 0)
	{
		printf("No K4A devices found\n");
		return 0;
	}

	if (K4A_RESULT_SUCCEEDED != k4a_device_open(K4A_DEVICE_DEFAULT, &device))
	{
		printf("Failed to open device\n");
		goto Exit;
	}

	config.color_format = K4A_IMAGE_FORMAT_COLOR_MJPG;
	config.color_resolution = K4A_COLOR_RESOLUTION_2160P;
	config.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;
	config.camera_fps = K4A_FRAMES_PER_SECOND_30;

	if (K4A_RESULT_SUCCEEDED != k4a_device_start_cameras(device, &config))
	{
		printf("Failed to start device\n");
		goto Exit;
	}

	while (captureFrameCount-- > 0)
	{
		k4a_image_t image;

		// Get a depth frame
		switch (k4a_device_get_capture(device, &capture, TIMEOUT_IN_MS))
		{
		case K4A_WAIT_RESULT_SUCCEEDED:
			break;
		case K4A_WAIT_RESULT_TIMEOUT:
			printf("Timed out waiting for a capture\n");
			continue;
			break;
		case K4A_WAIT_RESULT_FAILED:
			printf("Failed to read a capture\n");
			goto Exit;
		}

		printf("Capture");

		// Probe for a color image
		image = k4a_capture_get_color_image(capture);
		if (image)
		{
			printf(" | Color res:%4dx%4d stride:%5d ",
				k4a_image_get_height_pixels(image),
				k4a_image_get_width_pixels(image),
				k4a_image_get_stride_bytes(image));
			k4a_image_release(image);
		}
		else
		{
			printf(" | Color None                       ");
		}

		// probe for a IR16 image
		image = k4a_capture_get_ir_image(capture);
		if (image != NULL)
		{
			printf(" | Ir16 res:%4dx%4d stride:%5d ",
				k4a_image_get_height_pixels(image),
				k4a_image_get_width_pixels(image),
				k4a_image_get_stride_bytes(image));
			k4a_image_release(image);
		}
		else
		{
			printf(" | Ir16 None                       ");
		}

		// Probe for a depth16 image
		image = k4a_capture_get_depth_image(capture);
		if (image != NULL)
		{
			printf(" | Depth16 res:%4dx%4d stride:%5d\n",
				k4a_image_get_height_pixels(image),
				k4a_image_get_width_pixels(image),
				k4a_image_get_stride_bytes(image));
			k4a_image_release(image);
		}
		else
		{
			printf(" | Depth16 None\n");
		}

		// release capture
		k4a_capture_release(capture);
	}

	returnCode = 0;
Exit:
	if (device != NULL)
	{
		k4a_device_close(device);
	}

	return returnCode;
}`

output is:
[2019-08-05 18:55:57.250] [error] [t=9604] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (143): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   3151066 type:Depth
Capture | Color None                        | Ir16 res: 576x 640 stride: 1280  | Depth16 res: 576x 640 stride: 1280
[2019-08-05 18:55:57.283] [error] [t=9604] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (143): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   3184400 type:Depth
Capture | Color None                        | Ir16 res: 576x 640 stride: 1280  | Depth16 res: 576x 640 stride: 1280
[2019-08-05 18:55:57.316] [error] [t=9604] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (143): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   3217733 type:Depth
Capture | Color None                        | Ir16 res: 576x 640 stride: 1280  | Depth16 res: 576x 640 stride: 1280

Color is None
So I add "config.synchronized_images_only = true;"
But the result is:
[2019-08-05 18:58:10.735] [error] [t=10200] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (143): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   3617733 type:Depth
[2019-08-05 18:58:10.769] [error] [t=10200] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (143): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   3651066 type:Depth
[2019-08-05 18:58:10.802] [error] [t=10200] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (143): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   3684400 type:Depth
[2019-08-05 18:58:10.836] [error] [t=10200] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (143): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   3717733 type:Depth
[2019-08-05 18:58:10.869] [error] [t=10200] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (143): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   3751066 type:Depth
I can't get the depth data too....

Brent-A, 2019-08-05:
Hi @riverMountainjs, 

Can you verify two things:

**1. You can see the color camera using the Windows Camera app (or any other camera viewer application)**

**2. You can see both the depth and color using k4aviewer.**

You may want to try different configurations that are lower bandwidth (resolution and frame rate) to confirm that you aren't dropping frames due to a resource limitation.

riverMountainjs, 2019-08-06:
1. I can see the color camera camera using the Windows Camera app. I have used kinect v2.0, and I can see color images.
2. I can't open k4aviewer, it says:  Unable to locate program input point GetDpiForSystem in dynamic link library C:\ Program Files\Azure SDK v1.1.1\tools\k4aviewer.exe

Brent-A, 2019-08-06:
Interesting. What version of Windows are you running?

GetDpiForSystem was added in Windows 10, version 1607. Azure Kinect has a minimum OS version of [Windows 10 April 2018 release (x64)](https://docs.microsoft.com/en-us/azure/Kinect-dk/system-requirements). There may be other problems if you are running on an OS that is too old.

riverMountainjs, 2019-08-07:
Thanks！I updated the OS and the problem was solved.

closed 2019-08-07 by rabbitdaxi