microsoft/Azure-Kinect-Sensor-SDK #587: Incorrect image size calculation in k4a_image_create function for K4A_IMAGE_FORMAT_COLOR_NV12 format

bibigone, 2019-08-05:
Image size calculation inside `k4a_image_create` function is incorrect in case of `K4A_IMAGE_FORMAT_COLOR_NV12` format:

```c
k4a_image_create(K4A_IMAGE_FORMAT_COLOR_NV12, 1280, 720, 1280, &image);
```
creates image with buffer size `921600` bytes (I guess it is calculated as `sride * height` which gives `1280 * 720 = 921600` in our case).

But it is not correct size for NV12 image `1280x720`. Correct size must be `3 * stride * height / 2`  that is `1382400` bytes.

You can say that this is by design and I have to specify different stride for such NV12 image. But let's start color streaming from device in `K4A_IMAGE_FORMAT_COLOR_NV12` format and `K4A_COLOR_RESOLUTION_720P` resolution, then wait for the first capture and ask color image from it using `k4a_capture_get_color_image`. As a result we'll get image with the following parameters:
* format: ` K4A_IMAGE_FORMAT_COLOR_NV12`
* width: `1280` pixels
* height: `720` pixels
* stride: `1280` bytes (exactly as in our call to `k4a_image_create`)
* size: `1382400` bytes (but `k4a_image_create` returns image with size = `921600` bytes)

That is, at least we have some inconsistency between `k4a_image_create` and `k4a_capture_get_color_image` function for `K4A_IMAGE_FORMAT_COLOR_NV12` images.

SDK Version:  1.1.1

Brent-A, 2019-08-05:
Thanks @bibigone, unfortunately k4a_image_create makes a simplified assumption about the size of the image so it doesn't work for NV12. You can use k4a_image_create_from_buffer, although it's a bit more complex.

I think we need to introduce a new API (perhaps k4a_image_create_with_size) to handle the NV12 case without the user having to provide their own buffer.

bibigone, 2019-08-05:
> unfortunately k4a_image_create makes a simplified assumption about the size of the image so it doesn't work for NV12

I see. But there're at least two better solutions than simply create incorrect image:
1. Adding the following special case processing to image size calculation:
```c
if (format == K4A_IMAGE_FORMAT_COLOR_NV12) {
    size = 3 * size / 2;
}
```
2. Returning `K4A_RESULT_FAILED` from `k4a_image_create` if format is equal to `K4A_IMAGE_FORMAT_COLOR_NV12` (with additional message to log).

I would prefer the first solution, but the second one is also better than current behavior.

> I think we need to introduce a new API (perhaps k4a_image_create_with_size)

This is really good idea independent on issue with `K4A_IMAGE_FORMAT_COLOR_NV12` format.

PR #594 (2019-08-05): Made image_create format aware

## Fixes #587 

### Description of the changes:
- Made image_create stride aware

image_create will now validate the basic requirements of known formats, and allocate sufficient memory for the NV12 format.

### Before submitting a Pull Request:
- [X] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [X] I [built my changes](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) locally
- [X] I ran the [unit tests](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob [... cut]
closed 2019-08-05 by Brent-A