microsoft/Azure-Kinect-Sensor-SDK #588: Which filter method used in k4a_calibration_2d_to_3d when unproject a depth image to point cloud

YuXinFan, 2019-08-05:
The k4a kinect does a very good point cloud. At first, I think it is due to the good depth sensor. However, when I do unprojection by myself after getting the depth map, the result has lots of noise. I read the sdk source code and find that kinect use a filter method to do that. But the source code also does an undistortion at the same time. I am confused that I can't recognize the filter method. Therefore I wonder which filter method is used in kinect sdk.

Thanks.

rabbitdaxi, 2019-08-05:
@YuXinFan please check out this page: https://docs.microsoft.com/en-us/azure/Kinect-dk/use-image-transformation, recommend to use (k4a_transformation_depth_image_to_point_cloud) https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga7385eb4beb9d8892e8a88cf4feb3be70.html#ga7385eb4beb9d8892e8a88cf4feb3be70 to generate point cloud. The function implementation is in the SDK source code. You can also look at the transformation example: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/transformation if you are interested in color point cloud. 

The Azure Kinect viewer's source code: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/tools/k4aviewer is also a good reference of how the out of box visualization tool generate the point cloud (no post processing from viewer, straight render from the SDK)

YuXinFan, 2019-08-06:
@rabbitdaxi Thanks for your replay.   
Use **k4a_transformation_depth_image_to_point_cloud** is a good idea. But I have to do unproject out of the real-time k4a program. This function needs a parameter **k4a_calibration_t**. If I use **k4a_transformation_depth_image_to_point_cloud** function to do this, I need the struct *k4a_calibration_t*. But I have no idea about how to store **k4a_calibration_t** in a file so that another program can load the calibrations from the file.

rabbitdaxi, 2019-08-06:
@YuXinFan if you do not need live streaming, the API should be able to enable you to do playback as well. For example, you can use k4arecorder to record mkv and playback it. In the [transformation example](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/transformation), you still can see there is a playback option that you can get k4a_calibration_t from the mkv file. 

If you have your customized reason that need to store k4a_calibration_t as binary file and load the binary buffer somewhere else, that should also work. 

Please let us know whether we answered your question. If not, please provide details of what you are trying to do and what the current issue you are having (e.g. sample images). 

As a side note, if you are asking whether we have post processing from the 2d depth to 3d point cloud to filter noise, the answer is not really, you can look at the implementation of the k4a_transformation_depth_image_to_point_cloud, we basically just unproject the 2d depth to 3d with the brownconrady intrinsics model

YuXinFan, 2019-08-07:
@rabbitdaxi Thanks for your answer. But it does not really solve my problem. The problem I met is that after I get depth images and intrinsics and distortion parameters from the kinect dk and try to get the point cloud of the depth image, I apply undistortation on depth image using *cv::initDistortationMap* and *cv::remap* with parameters k1,k2,k3,k4. Then calculate 3d position as *x=(u-cx)/fx * d, y = (v-cy)/fy * d, z = d, also filter the points with zero positions. However, the result point cloud has lots of noise on the lines from the edge of objects to the camera focus, unlike the point cloud of sdk viewer tool.  

To solve this problem, I find the source code of sdk does an **iterative unproject** operation for each point when doing unproject,  which does re-projection and compares some error many times. It seems that this function is a filter to improve the result of unprojection. It likes compares re-projection and raw point errors, but I am not sure.

The source code of kinect viewer also uses this method to convert depth image to point cloud.

I think to apply this filter in the source code on my result may be much better. Therefore, I wonder what the function **transformation_iterative_unproject** really does. And dose my simply undistortation operation matter?

rabbitdaxi, 2019-08-07:
@YuXinFan Thank you for explaining what you are trying to do. I think I am able to provide some pointers to help you understand the calibration and geometry.

- **k4a_transformation_depth_image_to_point_cloud** is still my recommended method for generating point cloud from the depth image, especially we have LuT+SSE implementation of this function which has fast real-time performance. It **does not** contain any post filter processing. The **transformation_iterative_unproject* that compute the 2d to 3d ray LuT implements a common approach that finding correspondence from 2d to 3d with optimization on the re-projection error term (internally compute a Jacobian). This method is operating on the depth image straight from the SDK (distorted), and the project/unproject function internally take into account of the radial and tangential distortion parameters. 
- Regarding your question about all the intrinsics parameters meaning, please read [API Doc](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/structk4a__calibration__intrinsic__parameters__t_1_1__param.html), you also can find an simple example of [opencv compatibility code snippet](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/opencv_compatibility) that shows how k4a intrinsics and opencv can be connected. The camera model is BrownConrady which is one of supported model by OpenCV (projectpoints can be directly used) 
- Regarding what your current approach that try to undistort the depth image and then use the simple pinhole camera model to unproject to 3d. The noise could be introduced by your undistort function if you have the cv::remap contains the interpolation method with INTER_LINEAR, is that what you have? 

I still recommend you read our [documentation](https://docs.microsoft.com/en-us/azure/Kinect-dk/use-calibration-functions). And **k4a_transformation_depth_image_to_point_cloud** exactly what we provided to user to do such task.

YuXinFan, 2019-08-08:
@rabbitdaxi Thanks for your patience.  You are right that the noise is mostly caused by the interpolation method. I change *INTER_LINEAR* to *INTER_NEAREST*. Then the noise is disappeared.

However, the *INTER_NEAREST* method makes objects deformation. The result point cloud will lose shape information. For example, my calibration board after *INTER_NEAREST* undistortion:
[image]

I am confused that why not make undistortion an independent part so that we can freely to choose sdk function method with our own function method in each process. I feel these functions' source code implementation method makes it much coupling with other processes. 

The point cloud is not my aim. My existing program interface APIs don't want to be changed. I am almost to change the module of camera deriver from the old camera to Kinect. And I don't want this change influence other modules. But I feel hard to let Kinect fit the existing program.

rabbitdaxi, 2019-08-08:
@YuXinFan It makes sense that when you undistort with LINEAR which introduce the flying pixels on the edge (between depth with 0 value and non-0 value). Interpolation with NN will give you the aliasing artifacts as in your image. One method that can help is using bilinear method with handling the 0 depth pixels during computing weight. There is a sample that you can take a look as a reference. https://github.com/microsoft/Azure-Kinect-Samples/blob/master/opencv-kinfu-samples/main.cpp. You can look at the CUSTOM_UNDISTORTION section. Hope it helps

rabbitdaxi, 2019-08-09:
We will update the undistort example with more details.

closed 2019-08-12 by rabbitdaxi

YuXinFan, 2019-08-13:
@rabbitdaxi Thanks for your answers. Your answers help me a lot.

1106204321, 2019-08-29:
> the implementation of the k4a_transformation_depth_image_to_point_cloud
where could I find the  implementation of  the k4a_transformation_depth_image_to_point_cloud?

rabbitdaxi, 2019-09-05:
@1106204321 , the code is open sourced for this function, you can just search the function in k4a.cpp, which it will call into https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/src/transformation/transformation.c

1106204321, 2019-09-06:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/588#issuecomment-528482512
thank you very much!

1106204321, 2019-09-06:
> @1106204321 , the code is open sourced for this function, you can just search the function in k4a.cpp, which it will call into https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/src/transformation/transformation.c

thanks

1106204321, 2019-09-18:
@rabbitdaxi ，Does the SDK provide source code of 'k4a_calibration_2d_to_3d' ?  I couldn't find it.

LogWell, 2021-02-04:
How to obtain aligned, undistorted RGBD data, in this case the amount of calculation for **projection** and **unprojection** is very small.

LogWell, 2021-02-04:
@1106204321 Please ref:
    /Azure-Kinect-Sensor-SDK/include/k4ainternal/transformation.h
    /Azure-Kinect-Sensor-SDK/src/transformation/*