microsoft/Azure-Kinect-Sensor-SDK #597: can't run transformation example

CGev, 2019-08-06:
**Describe the bug**
I tried to run the **transformation example** provided by the sdk but it doesn't compile with errors

**To Reproduce**
1) Clone the lastest version of the sdk and I used the command "git submodule update --init --recursive" to obtain all the project dependences 
2) Generate the Examples project with Cmake
3) Open the Examples.sln and had the Azure Kinect DK package directly in Visual Studio 2017
4) Add some include/libs links in the project properties via Visual Studio 2017 

RESULT: the **enumerate example** works but I still have errors on the **transformation example**.

**Expected behavior**
Compile the transformation example from sdk without errors

**Logs**

Erreur	LNK2019	symbole externe non résolu _tjInitDecompress référencé dans la fonction "int __cdecl playback(char *,int,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >)" (?playback@@YAHPADHV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z)	transformation	C:\Users\ChloéGevrey\Documents\clone_kinect\Azure-Kinect-Sensor-SDK\examples\transformation\main.obj	1

Gravité	Code	Description	Projet	Fichier	Ligne	État de la suppression
Erreur	LNK2019	symbole externe non résolu _tjDecompress2 référencé dans la fonction "int __cdecl playback(char *,int,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >)" (?playback@@YAHPADHV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z)	transformation	C:\Users\ChloéGevrey\Documents\clone_kinect\Azure-Kinect-Sensor-SDK\examples\transformation\main.obj	1	

Erreur	LNK2019	symbole externe non résolu _tjDestroy référencé dans la fonction "int __cdecl playback(char *,int,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >)" (?playback@@YAHPADHV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@Z)	transformation	C:\Users\ChloéGevrey\Documents\clone_kinect\Azure-Kinect-Sensor-SDK\examples\transformation\main.obj	1	

Erreur	LNK1120	3 externes non résolus	transformation	C:\Users\ChloéGevrey\Documents\clone_kinect\Azure-Kinect-Sensor-SDK\examples\Debug\transformation.exe	1	

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS with Version: Windows 10
 - SDK Version:  1.2.0-alpha.10
 - Visual Studio Version:  Community 2017

cdedmonds, 2019-08-06:
The generated project files are tested in our CI builds whenever a new check-in is made. Can you give further detail on what modifications you made to include/libs in Visual Studio?

CGev, 2019-08-07:
When I compiled the project for the first time, I had some errors about undefined files as "turbojpeg.h" so I've added its path into the project property > C/C++ > Include 
And I did the same to include k4a/k4arecord folders. 

"turbojpeg.h" error: 
[image]

rabbitdaxi, 2019-08-07:
@CGev , recommend you to try the cmake .. -GNinja which is preferred build method described in this [page](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md). If you really want to use cmake to generate VS project to build, other than include the turbojpeg.h, you need to include the libjpeg-turbo as well. If you are using cmake to build as I described above, the libjpeg-turbo as a submodule will be automatically included and you should build success without issue (this is also how our official CI build build the example)

schultetwin1, 2019-08-12:
@CGev, from your original message it looks like you are using CMake to generate a msbuild files (vcxproj and sln files) and then opening them in Visual Studio to build them. This _should_ work.

In your second step you said

> 2. Generate the Examples project with Cmake

What command line to you run to do this? Or do you use Visual Studio to do this?

Also, can you share your version of Visual Studio with us?

forrestsatterfield, 2019-08-15:
I'm having the same problem. I noticed when I use cmake in Visual Studio 2017 to build the entire SDK a warning message says 
"CMake Warning at extern/libjpeg-turbo/CMakeLists.txt:15 (message):
NASM assembler not found - libjpeg-turbo performance may suffer .\GitHub\Azure-Kinect-Sensor-SDK\extern/libjpeg-turbo/CMakeLists.txt"

schultetwin1, 2019-08-15:
@forrestsatterfield thanks for the info. The warning you are experiencing is expected. If you don't have the NASM assembler installed on your machine libjpeg-turbo will not have the same performance. However, that should prevent you from building.

Are you seeing the same build errors as CGev? If you aren't sure (since CGev's post is in French) feel free to open up a new issue with your build errors and we'll mark that issue as a duplicate of this one.

forrestsatterfield, 2019-08-15:
@schultetwin1 Yes I am experiencing the same error but in English.

schultetwin1, 2019-08-15:
I misunderstood the bug to begin with. My apologies.

You are both trying to open the Examples.sln file and build that. Is that correct?

That is an out of date solution file which is no longer maintained. I would suggest using CMake to build the examples and not MSBuild. Note, if you want to use Visual Studio, Visual Studio now [supports CMake](https://docs.microsoft.com/en-us/cpp/build/cmake-projects-in-visual-studio?view=vs-2017) and a build system.

As you can see in the linked PR, we are going to remove the msbuild files.

forrestsatterfield, 2019-08-16:
I figured that might be the case which is a shame! I'm a biomedical engineer using the azure Kinect for 3D scanning someone for a 3D printed knee brace and then using body tracking to assess the effect on gate. The transformation example looked like it would be really helpful for determining best 3D scanning method.

schultetwin1, 2019-08-16:
You should still be able to build the transformation example using CMake. It's just not able to build using the Examples.sln file. See [building.md](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) for instructions on how to do that.

closed 2019-08-16 by schultetwin1

nosignal00, 2020-10-26:
For whom someone visit this issue in the future,
The issue raised by @CGev is because of tjInitDecompress method from libjpeg-turbo, not cmake compile related settings.
The transformation example uses libjpeg-turbo and this one should be avaialble also as build file(like .lib, .dll).

The issues related to the cmake mentioned on this repo should be addressed from two perspectives, but only from one perspective.

The first one is that users build all components of SDK to use any function of in this SDK.
You can see @UnaNancyOwen's comment about this kind of situation on following thread.
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1212

But I think almost issue raisers about example code does not want to do that.

So, the second perspective, which is not well considered in this repo, is more appropriate for the developers 
who only want to use examples.
The second perspective is that users build only one example at a time.
So, I think if the Cmakelist.txt file is offered as stand alone for each example, it would be great for those users.

Anyway, for the transformation example case you can build as follows on Windows 10 without building all components.
1 . Open VStudio
2 . File > New Project > Console App C++ Project
3 . Select x64 as Debug Mode
4 . Manage Nuget Package
5 . Browse > Type and select: “Microsoft.Azure.Kinect.Sensor v1.4.1” and Install it.
6 . add transformation example realted files(*.h, *.cpp) to the project
7 . build with Cmake the folder "Azure-Kinect-Sensor-SDK\extern\libjpeg-turbo\src"
8 . add directory "Azure-Kinect-Sensor-SDK\extern\libjpeg-turbo\src" to the project
9 . add library files(*.lib, *.dll) from 7th step to the project
[image]
10. build, then you can see the result file in x64/Debug folder.

Good luck.

PR #649 (2019-08-16): Removes examples msbuild files
## Fixes #597 

### Description of the changes:
There are legacy msbuild files to build the examples which have not
been maintained. The examples build under CMake but not under these
msbuild files. Instead of maintaining two build systems we are
removing the msbuild files.

### Before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [x] I [built my changes](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) locally
- [ ] I ran the [u [... cut]