microsoft/Azure-Kinect-Sensor-SDK #603: Body Tracking is inaccurate when arms moves fast and worse when the device is vertical

wqs111000, 2019-08-07:
**Describe the bug**

1.When the body moves arms  fast in front of the sensor, the arm joints of the body skeleton will  be separate with the arms of the depth image,as the figure below shows.
[image]
What is the reason? How to fix it? I tried it with a more powerful graphics card and it didn’t improve. How and when does  the DNN model output its estimated skeleton? 

2. When the device is vertical (that is the cameras are vertical alignment), body Tracking is almost unusable,except for some very standard postures. 
[image]

**To Reproduce**
1. Stand in front of the sensor.
2. Move arms fast.

And 
1. Rotate the Kinect device to be vertical 
2. Open the body tracking viewer

wqs111000, 2019-08-07:
Some issues like #514 and #521 are related to this issue.

yijiew, 2019-08-12:
Thanks for reporting this issue. We are planning to solve these two issues you mentioned in the next release. Please stay tuned.

yijiew, 2019-08-12:
There is another issue that is opened regarding to the vertical mounting issue #604 . And we are planning to use #514 to track the latency issue. I will close this issue for duplication.

closed 2019-08-12 by yijiew

fractalfantasy, 2020-03-19:
I’ve created a feature request to address the latency, inaccuracy and slow speed of Azure Kinect body tracking by implementing the Kinect 2 ‘random forest’ model as an option.. pls upvote:

https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/39945454-legacy-body-tracking-like-kinect-v2

cc: @wqs111000