microsoft/Azure-Kinect-Sensor-SDK #604: Can I use kinect vertically for body tracking?

isyoon0721, 2019-08-07:
Hi, I use Kinect vertically.

Body tracking is relatively unstable compared to when used in the horizontal direction.

Do I need to replace variables to use vertically?

**Describe the bug**

Body Tracking unusable.

**To Reproduce**

Tested in Simple_3D_Viewer sample : WFOV 2x2 binned Mode

**Expected behavior**

**Logs**

**Screenshots**

[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS with Version: Win 10
 - SDK Version:  Azure Kinect Body Tracking SDK 0.9 

**Additional context**

yijiew, 2019-08-12:
Thanks for reporting this issue. The vertical mounting is not supported in the previous version of SDK, To address this issue, we will add an option for the customer to specify the mounting direction of the sensor when creating the tracker.

bibigone, 2019-08-14:
> To address this issue, we will add an option for the customer to specify the mounting direction of the sensor when creating the tracker.

It would be really great! Brilliant idea.

yijiew, 2019-08-30:
The new release 0.9.2 is published! The problem is resolved. You can download this now at: [msi](https://www.microsoft.com/en-us/download/details.aspx?id=100128),  [nuget](https://www.nuget.org/packages/Microsoft.Azure.Kinect.BodyTracking/0.9.2) or apt-get on Linux.

closed 2019-08-30 by yijiew