microsoft/Azure-Kinect-Sensor-SDK #605: New body tracker release, 100% CPU usage

nicolascarre, 2019-08-07:
Thanks for the new release, it seems to works better.
But now the CPU usage grow up to 100% immediately, even with the kinect azure body tracker sample, is that normal ?

RoseFlunder, 2019-08-09:
I observe the same behaviour after updating to version 0.9.1 of the Body Tracking SDK.
Here are my specs:

OS: Windows 10 1803
CPU: Intel Core i7-7800X
GPU: NVidia GTX 1080
RAM: 16GB

With a single kinect the body tracker sample takes all free CPU resources.
Can also confirm that its not the GUI of the samples because in other body tracking samples without GUI the CPU usage goes up to 100% too.
If I use two kinects for bodytracking simultaneously on this PC it still seems to work reasonably fast and CPU resources get split between the two processes.

yijiew, 2019-08-12:
Thanks for reporting this issue. We digged down in the SDK and found out that the bug was from the latest OnnxRuntime. We started a conversation with the ONNX team and they started investigate the issue. We will fix this issue in the next release.

yijiew, 2019-08-30:
The new release 0.9.2 is published! The problem is resolved. You can download this now at: [msi](https://www.microsoft.com/en-us/download/details.aspx?id=100128),  [nuget](https://www.nuget.org/packages/Microsoft.Azure.Kinect.BodyTracking/0.9.2) or apt-get on Linux.

closed 2019-08-30 by yijiew