microsoft/Azure-Kinect-Sensor-SDK #608: Publish SDK 1.2.0

Brent-A, 2019-08-07:
1.2.0 has reached beta, and is planned for official release on the week of Aug 19th.

**MSI Link:** *TBD will be posted soon*
**Nuget Link:** [1.2.0-beta.0](https://www.nuget.org/packages/Microsoft.Azure.Kinect.Sensor/1.2.0-beta.0)
**Deb package link:** *TBD will be posted soon*

**For anyone interested in trying the beta:**
Please download and try the 1.2.0 beta packages. File new issues if you encounter any problems, please clearly state the version of the SDK you encountered the problems with (e.g. 1.2.0-beta.0). If you have any other comments on the release, you can also comment on this issue.

**Firmware update:**
There is a new firmware file included in this release. Please follow the [firmware update instructions](https://docs.microsoft.com/en-us/azure/Kinect-dk/update-device-firmware) to update your firmware.

**Summary of [changes](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CHANGELOG.md) from 1.1.x:**
* C# (and other .NET language) support added. Compatible with .NET Standard 2.0.
* Depth Engine breaking changes to 2.0, SDK now relies on this new version.
* Firmware updates for better USB compatibility.
* Added new API's k4a_image_get_device_timestamp_usec(), k4a_image_get_system_timestamp_usec(), 
k4a_image_set_device_timestamp_usec(), k4a_image_set_system_timestamp_usec(), and k4a_image_set_exposure_usec().
* Deprecated API's k4a_image_get_timestamp_usec(), k4a_image_set_timestamp_usec(), and k4a_image_set_exposure_time_usec().
* Added new transformation API k4a_transformation_depth_image_to_color_camera_custom().
* Fixed color exposure get & set API's.
* The C++ API for playback was made public.
* Depth engine is now searched for in same the path as k4a.dll

See issues in this [milestone](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/milestone/3?closed=1).

initialneil, 2019-08-09:
@Brent-A Thanks for the update! With the new API, when using multiple cameras wire-synced, which timestamp should use to decide the same frame? `k4a_image_get_device_timestamp_usec` or `k4a_image_get_system_timestamp_usec`?

wes-b, 2019-08-09:
> With the new API, when using multiple cameras wire-synced, which timestamp should use to decide the same frame? k4a_image_get_device_timestamp_usec or k4a_image_get_system_timestamp_usec?

```k4a_image_get_device_timestamp_usec``` is the API returning the hardware timestamp used for device synchronization.

ichlubna, 2019-08-12:
Thanks, any way to get `libdepthengine.so.2.0` for Linux?

rfilkov, 2019-08-12:
> With the new API, when using multiple cameras wire-synced, which timestamp should use to decide the same frame? k4a_image_get_device_timestamp_usec or k4a_image_get_system_timestamp_usec? [...]

@wes_b, what is the idea behind the two timestamps? What is the use case, when should we use k4a_image_get_system_timestamp_usec then?

rfilkov, 2019-08-12:
@Brent-A Is there msi of 1.2.0b0 available already?

tesych, 2019-08-12:
@rfilkov, the msi is not available for 1.2 beta.

jrterven, 2019-08-14:
How to install the nuget package in Ubuntu 18.04?
I already installed mono-devel and nuget on linux as described here: https://docs.microsoft.com/en-us/nuget/install-nuget-client-tools
What's next?

venscn, 2019-08-16:
If I has many custom image, Can the parameter transformed_depth_image of this function k4a_transformation_depth_image_to_color_camera_custom be null?

wes-b, 2019-08-19:
> @wes_b What is the idea behind the two timestamps? What is the use case, when should we use k4a_image_get_system_timestamp_usec then?

@rfilkov, k4a_image_get_system_time_nsec() is a time measured by the host system using QueryPerformanceCounter / clock_gettime(CLOCK_MONOTONIC) to capture the host system time when the image arrived. k4a_image_get_device_time_usec is the device's time stamp (that can be synchronized across multiple devices) and has to mapping to host system time.

venscn, 2019-08-20:
@wes-b
This will waste a lot of memory and time to handle many custom image.

christianmakela, 2019-08-20:
@venscn 
The depth output buffer is needed for checking occlusions. If client passed in a null pointer then the SDK would have to allocate a temporary buffer for storing the output depth.

The client can reuse the same depth output buffer for all calls to k4a_transformation_depth_image_to_color_camera_custom() which means no memory is wasted vs. calling k4a_transformation_depth_image_to_color_camera_custom() just once.

jbrownkramer, 2019-08-20:
We've tried out the beta.  It drops in just fine to our system.  Do you know when it is likely to come out of beta?

venscn, 2019-08-22:
@christianmakela Thank u

wes-b, 2019-08-23:
MSI and NuGet have been published. The Debian is delayed until next week unfortunately.

http://download.microsoft.com/download/1/9/8/198048e8-63f2-45c6-8f96-1fd541d1b4bc/Azure%20Kinect%20SDK%201.2.0.msi

closed 2019-08-23 by wes-b

fishcu, 2019-08-28:
> How to install the nuget package in Ubuntu 18.04?
> I already installed mono-devel and nuget on linux as described here: https://docs.microsoft.com/en-us/nuget/install-nuget-client-tools
> What's next?

Run `mono nuget.exe install Microsoft.Azure.Kinect.Sensor` in the folder where you want to download the package(s). You will find `libdepthengine.so.2.0` in the `Microsoft.Azure.Kinect.Sensor.1.2.0/linux/lib/native/x64/release/` subfolder.

simplecosmas, 2019-09-25:
I cannot install this Nuget Link: 1.2.0, on my visual studio for windows get this error "Azure kinect only supports the x86/x64 platform" and my computer is a x64 bit platform.

wes-b, 2019-10-01:
> I cannot install this Nuget Link: 1.2.0, on my visual studio for windows get this error "Azure kinect only supports the x86/x64 platform" and my computer is a x64 bit platform.

If you are still hitting this, then please open a new issue, and use the bug template share more info.  Your Windows and VS versions would be super helpful.

PR #670 (2019-08-23): Publishing 1.2.0
## Fixes #608

### Description of the changes:
- Adding 1.2.0 MSI, 
- Changing SDK version to 1.3, and
- Using the non-alpha Nuget Package for develop

### Before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [x] I [built my changes](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) locally
- [ ] I ran the [unit tests](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/testing.md)
- [ ] I ran the [functional tests](https://g [... cut]