microsoft/Azure-Kinect-Sensor-SDK #611: Wrong values of *_track_enabled fields in k4a_record_configuration_t structure

bibigone, 2019-08-08:
I'm not sure is it bug in internals of `k4a` library or is it bug in `k4arecorder.exe` tool from Sensor SDK  but I have the following issue:
* When I record some MKV video using `k4arecorder.exe` tool with depth, color, IR and IMU streams
* And then open it via `k4a_playback_open`
* And ask configuration using `k4a_playback_get_record_configuration`
* In result `k4a_record_configuration_t` structure I see the following values:
  * `color_track_enabled` == **true**  // correct value
  * `depth_track_enabled` == **false** // *wrong* value! (recording actually has depth track)
  * `ir_track_enabled` == **false** // *wrong* value! (recording actually has IR track)
  * `imu_track_enabled` == **false** // *wrong* value! (recording actually has IMU track)
* Then, to be sure that all tracks do exist in recording, I read captures and IMU samples from it, using appropriate API, and all data is actually there. 

I've tried 1.1.0 and 1.1.1 versions of Sensor SDK to record and to read recordings. Issue persists in both versions and in any combination of them (for example: recording using 1.1.0, but reading in 1.1.1 and vice-versa). OS: Windows 10

rfilkov, 2019-08-08:
I have the same issue, as reported earlier here: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/339#issuecomment-502819239

xthexder, 2019-08-08:
If you open the recording with the Azure Kinect Viewer, installed via the MSI, what do you see under `Recording Settings` ?

Are you also seeing this same behavior when opening one of the sample recordings from #339, or just with your own recordings?

bibigone, 2019-08-09:
> If you open the recording with the Azure Kinect Viewer, installed via the MSI, what do you see under Recording Settings?

I see correct information. There is no values of `color_track_enabled`, `depth_track_enabled` and `ir_track_enabled` there (only modes and resolutions and they are correct in `k4a_record_configuration_t` structure also), but IMU Enabled = TRUE in UI while I have FALSE value in `imu_track_enabled` field of `k4a_record_configuration_t` structure.

> Are you also seeing this same behavior when opening one of the sample recordings from #339, or just with your own recordings?

Yep, I have exactly the same behavior with recordings from #339: correct information in Azure Kinect Viewer from Sensor SDK 1.1.1, but FALSE values in `depth_track_enabled` and `ir_track_enabled` and `imu_track_enabled` fields of `k4a_record_configuration_t` structure returned by `k4a_playback_get_record_configuration` function.

bibigone, 2019-08-09:
Here you can find log generated during calls to `k4a_playback_open` and `k4a_playback_get_record_configuration` but before captures reading from it:
[k4a_playback_open.log](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/3485897/k4a_playback_open.log)

rfilkov, 2019-08-09:
> I see correct information. There is no values of color_track_enabled, depth_track_enabled and ir_track_enabled there (only modes and resolutions and they are correct in k4a_record_configuration_t structure also), but IMU Enabled = Yes in UI while I have FALSE value in imu_track_enabled field of k4a_record_configuration_t structure.
> 
> Yep, I have exactly the same behavior with recordings from #339: correct information in Azure Kinect Viewer from Sensor SDK 1.1.1, but FALSE values in depth_track_enabled and ir_track_enabled and imu_track_enabled fields of k4a_record_configuration_t structure returned by k4a_playback_get_record_configuration function.

Ditto.

bibigone, 2019-08-09:
Finally, I got it: size of `bool` is compiler dependent and doesn't specified in standard, but most of modern compilers actually use 1 byte for `bool`. In my case it was 1 word... As a result, fields in structure `k4a_record_configuration_t` was differently aligned (in comparison with compiled version of k4a).

I think, you can close this issue as *invalid*. Sorry for bothering you...

But it would be great to have somewhere in documentation short note about alignment of fields in structures.

wes-b, 2019-08-09:
Thanks for the update @bibigone, can you help me understand better how you hit this? Are you using a different compiler. Wondering what we can do to help avoid this in the future.

xthexder, 2019-08-09:
This is definitely a valid issue. This structure should work across different compilers, but it currently doesn't.
It would be good to know which compiler you're using.

The reason the viewer works is because it was compiled with the same compiler as the SDK dll. Since this struct goes across a DLL boundary it should have `#pragma pack` set so the size doesn't change.
This is currently done for some structs that are serialized to file, but this struct is missing a packing definition.

bibigone, 2019-08-09:
> can you help me understand better how you hit this?

I wrote the simplest C code and compiled it using Visual Studio 2017. It worked correctly. After that I printed and compared `sizeof(k4a_record_configuration_t)`

bibigone, 2019-08-09:
> It would be good to know which compiler you're using.

Actually, it was C#... I intentionally omitted this detail in original issue description to keep things simple and clear. I was sure of correctness of my C# code... But I was wrong. While `sizeof(bool) == 1` in C#, for marshaling 4 byte values are used for `bool` (suppose, to be compatible with `BOOL` type in Windows API).

Sorry, that I didn't mention it in description. Next time, I'll try to reproduce issue on pure C before reporting.

wes-b, 2019-08-09:
This sounds more like a marshalling issue rather than a compiler issue. We know that MS, GCC, and Clange compilers are consistent. So for now I am going to close this issue.

closed 2019-08-09 by wes-b