microsoft/Azure-Kinect-Sensor-SDK #617: C# wrapper missing Custom transformation API

Brent-A, 2019-08-08:
**Describe the bug**
The new custom transformation APIs were added to C and C++, but not yet to C#.

rabbitdaxi, 2019-08-30:
Other than k4a_transformation_depth_image_to_color_camera_custom(), there is another newly added one k4a_calibration_color_2d_to_depth_2d(), they both in C\C++, not in C#

UnaNancyOwen, 2020-02-27:
I think this is required function. Body index map image can't be transformed using C# wrapper.
When will it be implemented? (Do you need to set milestone to [Release 1.4.0](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/milestone/10)?)
Thanks,

UnaNancyOwen, 2020-03-02:
@Brent-A @rabbitdaxi I have sent a pull request to fix this issue. Please review it! Thanks,

PR #1097 (2020-03-10): Add missing functions to C# wrapper

## Fixes #617

### Description of the changes:
- This pull request will add thease missing functions to C# wrapper.
  - Microsoft.Azure.Kinect.Sensor.DepthImageToColorCameraCustom()
    This function wrapped k4a_transformation_depth_image_to_color_camera_custom().
  - Microsoft.Azure.Kinect.Sensor.Calibration.TransformColor2DToDepth2d()
    This function wrapped k4a_calibration_color_2d_to_depth_2d().

### Before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [... cut]
closed 2020-03-10 by MsDerekMa