microsoft/Azure-Kinect-Sensor-SDK #620: Why black hair noise issue more serious in WFOV mode  than in NFOV mode ?

carlhuang070, 2019-08-09:
Why black hair noise issue more serious in WFOV mode  than in NFOV mode ?
Is there any way to solve hair noise  from top down Capture depth image?

rabbitdaxi, 2019-08-09:
@carlhuang070 , the depth camera is a Time of flight camera, therefore, low reflectivity object tends to have more noise simply because the return signal is less compare to high reflectivity objects at the same distance. 

The reason WFOV mode contains more noise on such low reflectivity objects is because the signal strength received by sensor is less than NFOV per pixel. The Azure Kinect is a USB powered device, with limited power resource, 4 different depth modes were designed for supporting different field of view and range to fulfill various users' needs. Refer to [this page](https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification#depth-camera-supported-operating-modes) to understand each mode's field of view and range.

carlhuang070, 2019-08-12:
@rabbitdaxi , Can we adjust the signal intensity to reduce hair noise in WFOV?

rabbitdaxi, 2019-08-12:
@carlhuang070 , the signal intensity  is a hardware configuration which is not adjustable from SDK (even though it is possible to configure illumination profile and gain from the depth camera module level, but in that case, we are talking about a new hardware). 

Are you using WFOV2X2BINNED or WFOVUNBINNED? the WFOV2X2BINNED should have stronger signal compare to the UNBINNED.

Other than that, your best choices here are either use NFOV modes, or move camera closer to the hair.

rfilkov, 2019-08-12:
@carlhuang070 According to my previous experience with Kinect-v2, a workaround to a similar issue was to place a light source in front and above the users' area.

carlhuang070, 2019-08-13:
@rabbitdaxi , i using WFOV2*2binned, the distance between DK and hair about 1.5m.
and the noise is very seriously.

carlhuang070, 2019-08-13:
@rabbitdaxi , I compare Kinect DK NFOV 2*2binned mode  with Kinect V2,  DK hair noise issue is more seriously at the same distance far away Camera . 
I found DK power: 5V ,2.5A
             Kinect V2: 12V,2.67A

rabbitdaxi, 2019-08-13:
@carlhuang070 thank your for your response. Your assumption that Azure Kinect consumes less power compare to Kinect V2 is correct. Azure Kinect is design for low power consumption (just USB powered), Kinect V2 require the AC power adapter. Your issue can not be solved by software, it is what the current DK hardware design offers. You can add your feedback to the feedback hub https://feedback.azure.com/forums/920053-azure-kinect-dk

closed 2019-08-13 by rabbitdaxi

zx-li001, 2019-11-28:
> @carlhuang070  as you siad, i am using a led light tube in front and above my head, but noise still the same as before. wondering what type of illumination light source you used?

zx-li001, 2019-11-28:
this problem seems will be solved when the kinect depth sensor has a HDR mode.