microsoft/Azure-Kinect-Sensor-SDK #622: Azure Kinect Viewer built from lastest code cannot find DLL depthengine_2_0

cybe-robot, 2019-08-09:
commit: 75d456cae16ae96eac8df14291c4dd58246568f7
As the depthengine.md pointed out, developers can get a depthengine_<major>_<minor>.dll from  a windows binary installer. Now, the situation is depthengine_1_0.dll can be access easily from version v1.1.1, v1.1.0 and v1.0.2, but I can get depthengine_2_0.dll, which is required by lastest version (commit hash code seen above) of Azure Kinect Viewer, from nowhere.  Hope the problem can be resolved as soon, sincerelly!

cybe-robot, 2019-08-09:
OK, here it is.
https://www.nuget.org/packages/Microsoft.Azure.Kinect.Sensor/1.2.0-alpha.10

closed 2019-08-09 by cybe-robot

wes-b, 2019-08-09:
Is there a better place would could have documented this temporary condition to make the Alpha release more discoverable?